ubopod/ubo_app #5: To have a hardware shortcut to start/restart ubo-app.

sassanh, 2023-11-30:
It just needs to somehow run `ubo` command, the less it depends on software the better.
Later we may have an `ubo_installer.sh` file which installs dependencies of ubo in OS and then installs ubo itself. People should be able to install ubo and all its dependencies in a standard system by just running a one-liner like this:
```sh
curl https://getubo.com/ubo_install.sh | sudo sh
```
And the hardware should have a shortcut to run it. The shortcut could be a complex one like "put your finger on light sensor and turn on and off the mic switch 3 times rapidly" or something like that. It is used rarely to save time for the user and avoid them being blocked when they have no ethernet cable nor a keyboard/mouse/display nor a sd-card reader.

This is to make ubo device easier to bootstrap and more self-sufficient.

mehrdadfeller, 2023-11-30:
I agree on having a way to automatically run this at startup. Overall user has two options, 

1. start with new a Raspberry Pi OS Image and run the shell script to install/run ubo app 
2. use the prepped images that will come with the device when they buy it. There's really no middle ground here since bootstrapping the install still requires some pre-installed dependencies and if the user has to install those, they might as well just install ubo-app. In other words, the short-cut still requires something installed.

One thing to mention however is that on the Top PCB there's an EEPROM that OS uses to load the device tree information. I am using that currently to store serial number and hardware test data as json. This can be used to automatically load device-tree info and certain preset. It might be possible to have the EEPROM flashed with some information to bootstrap the install but not so sure if and how this can be accomplished....specially since there is not internet at bootup before setting up WiFi etc...

sassanh, 2024-08-17:
@mehrdadfeller Just brainstorming material:

We can probably have an embedded sd-card interface with a readonly sd-card in it, like the recovery partition of the Apple computers.

When the user say holds a combination of keys, the system reboots to this recovery disc. It is done by making the exposed sd-card slot offline and the embedded slot online and rebooting the system.

When the os is loaded from the embedded sd-card, we can probably load the whole os + an image of ubo-app from embedded sd-card into memory and signal the hardware to make the embedded sd-card slot offline and the exposed slot online and then write the image into the user's sd-card.

We can also give the user some options in a menu, one option can be the one explained above (writing the whole image) while the other option can be chrooting into the os on the user's sd-card and installing ubo-app in it without erasing its data.