ubopod/ubo_app #16: Add audio feedback to WiFi onboarding process

ubopod, 2023-12-10:
Please refer to the previous implementation of WiFi onboarding for audio feedback chimes and messages for different phases of on-boading (success scan, success connecting to WiFi, etc)

mehrdadfeller, 2024-01-04:
one possible solution is to add `audio_chime` string field to notifications that can accepts an arbitrary string. If the string matches audio file name (for example `success.wav` for `success` string), the corresponding audio chime will be played with notifications. 

We can also add Text to Speech (TTS) to notifications to read notification message body out loud for accessibility. This can be a `True`/`False` flag.

mehrdadfeller, 2024-01-15:
I forgot to mention that to get audio working you have to first install the driver by running this script as `sudo`:

https://github.com/ubopod/ubo-sdk/blob/main/system/setup/install_wm8960.sh

and `reboot`.

You can then use `AudioMananger` as an example to play audio files:

https://github.com/ubopod/ubo-sdk/blob/main/audio/audio_manager.py

Please let me know if you run into any issues in getting audio working

commit fe953367 (2024-01-18): fix: #16 use a logarithmic scale for volume
commit 7313b145 (2024-01-19): feat: #16 add volume chime

fix: #16 set the default pulseaudio device to "wm8960-soundcard"
closed 2024-01-18 by sassanh