ubopod/ubo_app #17: remote vscode tunnel support

ubopod, 2023-12-11:
https://code.visualstudio.com/docs/remote/tunnels

sassanh, 2024-01-05:
We need the user to accept their license agreement somehow, I'm not sure if it's legally enough to show a qr code link to their agreement page with an accept menu item button?

ubopod, 2024-01-05:
You can run the command with a flag to automatically accept the license agreement for now. We can later do that through a popover with the yes/no options and qr code link to the agreement document...

mehrdadfeller, 2024-01-05:
"--accept-server-license-terms"

mehrdadfeller, 2024-04-19:
> @mehrdadfeller <https://github.com/mehrdadfeller> Yes I noticed that
> option but I guess we can't accept the term on behalf of the user. What do
> you think about showing a QRCode redirecting the user to their terms of
> service page with "accept" and "reject" buttons?
> We can make this widget reusable for other third party pieces of software.

Yes it makes sense.

We can have a dedicated widget for Terms of Service which would have three
buttons on the left hand side mapped to three actions:

- Read
- Accept
- Decline

When the user clicks on read button, we can show a QR code that they scan
and it will redirect them to the terms of service that they can read.

closed 2024-05-03 by sassanh