ubopod/ubo_app #24: headless_kivy_pi does not get installed when installing ubo-app with pip

mehrdadfeller, 2023-12-28:
I created a virtual environment on a fresh Pi OS and attempted to install and run ubo-app using `pip`:

```
ubo@ubo10:~ $ python -m venv ~/ubo --system-site-packages
ubo@ubo10:~ $ source ubo/bin/activate
```
Then I installed the app:

```
(ubo) ubo@ubo10:~ $ pip install ubo-app
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting ubo-app
  Downloading ubo_app-0.7.2-py3-none-any.whl (37 kB)
Collecting psutil<6.0.0,>=5.9.6
  Downloading psutil-5.9.7.tar.gz (498 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 498.4/498.4 kB 6.4 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting python-debouncer<0.2.0,>=0.1.0
  Downloading python_debouncer-0.1.1-py3-none-any.whl (7.9 kB)
Collecting python-redux<0.10.0,>=0.9.6
  Downloading python_redux-0.9.6-py3-none-any.whl (12 kB)
Collecting pythonping<2.0.0,>=1.1.4
  Downloading https://www.piwheels.org/simple/pythonping/pythonping-1.1.4-py3-none-any.whl (16 kB)
Collecting pyzbar<0.2.0,>=0.1.9
  Downloading pyzbar-0.1.9-py2.py3-none-any.whl (32 kB)
Collecting sdbus-networkmanager<3.0.0,>=2.0.0
  Downloading https://www.piwheels.org/simple/sdbus-networkmanager/sdbus_networkmanager-2.0.0-py3-none-any.whl (248 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 248.3/248.3 kB 505.6 kB/s eta 0:00:00
Collecting lock<2019.0.0.0,>=2018.3.25.2110
  Downloading https://www.piwheels.org/simple/lock/lock-2018.3.25.2110-py3-none-any.whl (5.1 kB)
Collecting python-immutable<2.0.0,>=1.0.0
  Downloading python_immutable-1.0.2-py3-none-any.whl (6.6 kB)
Collecting sdbus>=0.8rc2
  Downloading sdbus-0.11.1-cp37-abi3-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (535 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 535.6/535.6 kB 1.3 MB/s eta 0:00:00
Building wheels for collected packages: psutil
  Building wheel for psutil (pyproject.toml) ... done
  Created wheel for psutil: filename=psutil-5.9.7-cp311-abi3-linux_aarch64.whl size=282792 sha256=f0a104f4a014b4187d501d153b08342f4ea0afe38bd14d9fce1a66fb7feb7123
  Stored in directory: /home/ubo/.cache/pip/wheels/8a/e0/cf/0387173bc7d441888e75433d029f2d582ecc243b2f03835bc4
Successfully built psutil
Installing collected packages: pyzbar, pythonping, lock, sdbus, python-immutable, psutil, sdbus-networkmanager, python-redux, python-debouncer, ubo-app
  Attempting uninstall: psutil
    Found existing installation: psutil 5.9.4
    Not uninstalling psutil at /usr/lib/python3/dist-packages, outside environment /home/ubo/ubo
    Can't uninstall 'psutil'. No files were found to uninstall.
Successfully installed lock-2018.3.25.2110 psutil-5.9.7 python-debouncer-0.1.1 python-immutable-1.0.2 python-redux-0.9.6 pythonping-1.1.4 pyzbar-0.1.9 sdbus-0.11.1 sdbus-networkmanager-2.0.0 ubo-app-0.7.2
```
After installation, runnig `ubo` throw the following error:

```       
(ubo) ubo@ubo10:~ $ cd ubo/bin/
(ubo) ubo@ubo10:~/ubo/bin $ ubo 
Traceback (most recent call last):
  File "/home/ubo/ubo/bin/ubo", line 5, in <module>
    from ubo_app import main
  File "/home/ubo/ubo/lib/python3.11/site-packages/ubo_app/__init__.py", line 5, in <module>
    from headless_kivy_pi import HeadlessWidget
ModuleNotFoundError: No module named 'headless_kivy_pi'
```
I had to install `headless_kivy_pi` manually but it must install automically as depenency of `ubo-app`:

```
(ubo) ubo@ubo10:~/ubo/bin $ pip install headless_kivy_pi
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting headless_kivy_pi
  Downloading headless_kivy_pi-0.5.10-py3-none-any.whl (15 kB)
Collecting adafruit-circuitpython-aw9523<2.0.0,>=1.1.7
  Downloading https://www.piwheels.org/simple/adafruit-circuitpython-aw9523/adafruit_circuitpython_aw9523-1.1.9-py3-none-any.whl (6.7 kB)
[... 56 lines cut ...]
[INFO   ] [Logger      ] Purge log fired. Processing...
[INFO   ] [Logger      ] Purge finished!
[INFO   ] [Factory     ] 190 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2, img_pil (img_ffpyplayer ignored)
 Traceback (most recent call last):
   File "/home/ubo/ubo/bin/ubo", line 8, in <module>
     sys.exit(main())
              ^^^^^^
   File "/home/ubo/ubo/lib/python3.11/site-packages/ubo_app/__init__.py", line 29, in main
     from ubo_app.menu import MenuApp
   File "/home/ubo/ubo/lib/python3.11/site-packages/ubo_app/menu.py", line 4, in <module>
     from ubo_gui.app import UboApp
 ModuleNotFoundError: No module named 'ubo_gui'
```
I had to install this manually myself too:

```
(ubo) ubo@ubo10:~/ubo/bin $ pip install ubo-gui
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting ubo-gui
  Downloading https://www.piwheels.org/simple/ubo-gui/ubo_gui-0.6.4-py3-none-any.whl (3.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.5/3.5 MB 2.8 MB/s eta 0:00:00
Installing collected packages: ubo-gui
Successfully installed ubo-gui-0.6.4
```

sassanh, 2024-04-15:
One needs to install ubo-app with this command:

```bash
pip install ubo-app[default]
```

btw, it's not a normal library one might ever install with pip, it needs to be installed with its own installer and in the future we will provide a debian package.
At some point in the future we may be able to remove `[default]` extra, it is a workaround for poetry not letting a package have a dependency with a certain extra for development environment and without that extra for the production environment.

sassanh, 2024-04-28:
I think it is not relevant anymore as the app installs by its installer and it is not supposed to be installed directly via pip.

closed 2024-04-28 by sassanh