ubopod/ubo_app #27: add power off option

ubopod, 2024-01-04:
On home page, please add power off / reboot option. The display should clear and show powering off message before power off takes place.  This would be necessary for safe headless restart/power-off.

sassanh, 2024-01-04:
It also needs root access

mehrdadfeller, 2024-01-04:
This is currently implement in SDK code here:

https://github.com/ubopod/ubo-sdk/blob/main/device/safe-reboot-shutdown.service
https://github.com/ubopod/ubo-sdk/blob/main/device/safe_reboot_shutdown.py

sassanh, 2024-01-04:
> This is currently implement in SDK code here:

This is using `sudo` internally in a non interactive shell, so if sudo is set to take user’s password it will fail.
We can run a service with root user as we already talked about it but even then if we had lots of services running in root user we need to reconsider our approach, currently these tasks need root user:
1. `power off` and `reboot`
2. Connecting to, disconnecting from and forgetting a wireless connection
3. Something you mentioned earlier about LED ring if I’m not mistaken

I can think of these solutions:
1. Dedicated service for each task
2. A service to rule them all
3. Implement a password entry using qrcode so that user can actually sudo in our app whenever needed (I think it would be annoying)
4. Considering our app is a semi operating system, let it run with root access, cause there is potential for more tasks needing root access in the future 

I think 2 and 4 are better options for long run.

closed 2024-03-19 by sassanh