ubopod/ubo_app #48: Screen reader prompt - disabling issue

mehrdadfeller, 2024-03-09:
When the device boots up a voice message keeps saying `To install the screen reader press control alt space.`

Since no keyboard is attached there is not way to silence this. As a temporary fix, we can just disable this and meanwhile we can open an upstream issue to have this only enabled when a keyboard is attached.

mehrdadfeller, 2024-03-29:
@sassanh I managed to silence the prompt by uninstalling `orca`. Not the best solution, but we can take this route while we figure out a more elegant solution such as turning it on/off via keypad actions.

```
sudo apt remove orca
``` 

Can you please add this to the build process?

sassanh, 2024-03-30:
I added this command to `install.sh`. Please check it in the latest image 0.12.0.

ubopod, 2024-03-30:
This issue still persists and orca is till installed:

```
_tmp_0@raspberrypi:~ $ sudo apt show orca
Package: orca
Version: 45.2-1~bpo12+rpt2
Priority: optional
Section: gnome
Maintainer: Debian Accessibility Team <pkg-a11y-devel@alioth-lists.debian.net>
Installed-Size: 15.1 MB
Depends: dconf-gsettings-backend | gsettings-backend, python3:any, gir1.2-glib-2.0, gir1.2-gtk-3.0, gir1.2-pango-1.0, gir1.2-wnck-3.0, gir1.2-gstreamer-1.0, gstreamer1.0-plugins-good, python3-brlapi, python3-cairo, python3-gi, python3-louis, python3-pyatspi, python3-speechd, speech-dispatcher, gsettings-desktop-schemas, xkbset
Recommends: xbrlapi
Suggests: brltty
Homepage: https://wiki.gnome.org/Projects/Orca
Download-Size: 2,110 kB
APT-Sources: http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
Description: Scriptable screen reader
 A flexible, extensible, and powerful assistive technology that
 provides end-user access to applications and toolkits that support the
 AT-SPI (e.g., the GNOME desktop).
 .
 Orca defines a set of default behaviors (reactions to application events) and
 key bindings (reaction to user key presses). These default behaviors and key
 bindings can be overwritten on a per-application basis. Orca creates a script
 object for each running application, which merges both the default behaviors
 and key bindings, and the application specific ones. Orca provides the
 infrastructure to activate and deactivate scripts, as well as a host of
 services accessible from within the scripts.

N: There is 1 additional record. Please use the '-a' switch to see it
```

mehrdadfeller, 2024-03-30:
@sassanh okay I understand what is happening: `orca` is the screen reader software but the prompt message comes from `piwiz`. If we disable `piwiz` it might have unintended consequences. The issue is discussed here:

https://raspberrypi.stackexchange.com/questions/118911/how-do-i-stop-the-audio-message-to-install-the-screen-reader-press-control-alt

At this point, I don't know what is the right way to handle this...

sassanh, 2024-04-15:
@mehrdadfeller I completely disabled userconfig service by default in the build process, one may decide to enable it or run it a single time later if they need it.

closed 2024-04-15 by sassanh