ubopod/ubo_app #53: audio breaking on ubo-app v0.11.6

mehrdadfeller, 2024-03-28:
This issue is happening on ubo-app v0.11.6 and the previous version v0.11.5
```
1711585629.120300 [ERROR] Something went wrong while playing an audio file
Traceback (most recent call last):
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/000-sound/audio_manager.py", line 117, in play
    output_device_index=self.find_respeaker_index(),
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/000-sound/audio_manager.py", line 78, in find_respeaker_index
    raise ValueError(msg)
ValueError: ReSpeaker for default device not found
```
It seems like that the audio driver is not loaded. `aplay` can't find it either:

```
ubo@raspberrypi:/opt/ubo $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
```

it seems like that driver is either not installed or loaded:

```
ubo@raspberrypi:/opt/ubo $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: 10 -- -- -- -- -- -- -- -- -- 1a -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- 58 -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --  
```

`1a` is the address of the audio chip. When the driver is properly loaded, this shows up as `UU`

This is odd since there's no change in the kernel version. Maybe the install script is not being executed properly or some permission issue?

```
ubo@raspberrypi:/opt/ubo $ uname -a
Linux raspberrypi 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
```

mehrdadfeller, 2024-03-28:
another by product of this issue is that when trying to start the WiFi process, the attempt to play an audio instruction fails with this uncaught exception and it breaks the whole flow. 

I looked at the logs in sentry and did not see this exception being reported though. do you know why?

```
1711586239.427695 [INFO] Opening audio file for playback - extra: {
    "filename_": "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/000-sound/sounds/scan.wav"
}
1711586239.439854 [ERROR] Something went wrong while playing an audio file
Traceback (most recent call last):
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/000-sound/audio_manager.py", line 117, in play
    output_device_index=self.find_respeaker_index(),
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/000-sound/audio_manager.py", line 78, in find_respeaker_index
    raise ValueError(msg)
ValueError: ReSpeaker for default device not found
```

ubopod, 2024-03-28:
Also I realized that since we are blacklisting the built-in audio kernel module, the following device should not appear on `aplay` but it does:

```
card 2: Headphones [bcm2835 Headphones]
```

sassanh, 2024-03-28:
This is happening because of this change: https://forums.raspberrypi.com/viewtopic.php?p=2187427&sid=a8be96266e99e34c1d83a99e2f2099c4#p2187427
We need to fix it upstream in https://github.com/waveshare/WM8960-Audio-HAT

mehrdadfeller, 2024-03-28:
I don't think this is related to this change since our installer script is already trying to modify `/boot/firmware/config.txt` and NOT `/boot/config.txt` as you check [here](https://github.com/ubopod/ubo-app/blob/main/ubo_app/system/install_wm8960.sh)

We have not changed the base image for many months now and the driver was being installed on previous releases prior to `v.0.11.5` and audio was indeed working. Also, the WM8960 has not changed for last 3 years. 

Our installer script is correctly making changes to `/boot/firmware/config.txt`. 

I believe this is some sort of race condition during building the image.

ubopod, 2024-03-29:
@sassanh The audio issue got fixed after I reinstalled the audio driver. I did not change anything and I just reinstalled the driver. I think this an issue related to the build system.

```root@raspberrypi:/home/_tmp_3# cd /opt/ubo/env/lib/python3.11/site-packages/ubo_app/system
root@raspberrypi:/opt/ubo/env/lib/python3.11/site-packages/ubo_app/system# bash install_wm8960.sh
Cloning into 'WM8960-Audio-HAT'...
remote: Enumerating objects: 79, done.
remote: Counting objects: 100% (41/41), done.
remote: Compressing objects: 100% (17/17), done.
remote: Total 79 (delta 29), reused 25 (delta 24), pack-reused 38
Receiving objects: 100% (79/79), 62.96 KiB | 741.00 KiB/s, done.
Resolving deltas: 100% (38/38), done.
Created symlink /etc/systemd/system/sysinit.target.wants/wm8960-soundcard.service → /lib/systemd/system/wm8960-soundcard.service.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
```

sassanh, 2024-03-29:
I see, this is the relevant part in build logs:
https://github.com/ubopod/ubo-app/actions/runs/8460684596/job/23179353095#step:9:1760

I think probably the opposite has happened here, the logs say "can't read /boot/firmware/config.txt", maybe we need to update our images to the latest, maybe a tool in our stack has upgraded with this change and is not working with older images, I need to investigate further.

sassanh, 2024-03-30:
This should be resolved in the latest builds, haven't released yet, you can download the images from here:
https://github.com/ubopod/ubo-app/actions/runs/8489662777
I've tested it and it worked for me.

ubopod, 2024-03-30:
I am cusrious what was the root cause of this issue?

sassanh, 2024-03-30:
> I am cusrious what was the root cause of this issue?

It would help if you try older images and see if they really had audio.

ubopod, 2024-03-30:
@sassanh issue still persists in version `0.12.0` of the release.

sassanh, 2024-04-18:
@mehrdadfeller It should be fixed in these images: https://github.com/ubopod/ubo-app/actions/runs/8743546506
I checked the lite image and it worked for me.

closed 2024-04-24 by sassanh