ubopod/ubo_app #54: docker start on GUI is not doing anything (v0.11.6)

mehrdadfeller, 2024-03-28:
When I press docker start, the screen tries to show something quickly but nothing happens after that. I don't see any errors in `ubo-app.log`. I only see the command is send in `system-manager.log`:

```
ubo@raspberrypi:/opt/ubo $ tail -f system-manager.log

1711586657.486268 [DEBUG] New connection: - extra: {
    "client_address": ""
}
1711586657.487705 [DEBUG] Received command: - extra: {
    "command": "b'docker start'"
}
```

sassanh, 2024-03-30:
@mehrdadfeller Please test it again in 12.0.0, it seems to be working for me.

sassanh, 2024-03-30:
I just ran Portainer and connected to it without any issues, I also stopped and started docker service via our GUI several times, I think this issue is resolved by itself.

closed 2024-03-30 by sassanh