ubopod/ubo_app #56: Camera fails to open during WiFi on-boarding

mehrdadfeller, 2024-03-29:
This is the log captured after I press the scan button during the flow:

```
1711694713.554487 [ERROR] Error loading "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/040-camera"
Traceback (most recent call last):
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/load_services.py", line 173, in run
    self.spec.loader.exec_module(self.module)
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/load_services.py", line 52, in exec_module
    super().exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/040-camera/ubo_handle.py", line 7, in <module>
    register_service(
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/load_services.py", line 253, in register_service
    result = init()
             ^^^^^^
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/040-camera/setup.py", line 80, in init_service
    from picamera2 import Picamera2  # pyright: ignore [reportMissingImports]
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/__init__.py", line 9, in <module>
    from .picamera2 import Picamera2, Preview
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 26, in <module>
    from picamera2.encoders import Encoder, H264Encoder, MJPEGEncoder, Quality
  File "/usr/lib/python3/dist-packages/picamera2/encoders/__init__.py", line 21, in <module>
    from .h264_encoder import H264Encoder
  File "/usr/lib/python3/dist-packages/picamera2/encoders/h264_encoder.py", line 3, in <module>
    from v4l2 import (V4L2_CID_MPEG_VIDEO_H264_I_PERIOD,
ImportError: cannot import name 'V4L2_CID_MPEG_VIDEO_H264_PROFILE' from 'v4l2' (/usr/lib/python3/dist-packages/v4l2.py)
1711694755.367959 [INFO] Opening audio file for playback - extra: {
    "filename_": "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/000-sound/sounds/done.wav"
}
1711695044.235809 [INFO] Opening audio file for playback - extra: {
    "filename_": "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/000-sound/sounds/scan.wav"
}
```

mehrdadfeller, 2024-03-29:
@sassanh I managed to fix this by upgrading the pip package; Can you please include this change in the next release?

```
(env) ubo@raspberrypi:/opt/ubo $ pip install --upgrade v4l2-python3
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: v4l2-python3 in /usr/lib/python3/dist-packages (0.3.2)
Collecting v4l2-python3
  Downloading https://www.piwheels.org/simple/v4l2-python3/v4l2_python3-0.3.4-py3-none-any.whl (22 kB)
Installing collected packages: v4l2-python3
  Attempting uninstall: v4l2-python3
    Found existing installation: v4l2-python3 0.3.2
    Not uninstalling v4l2-python3 at /usr/lib/python3/dist-packages, outside environment /opt/ubo/env
    Can't uninstall 'v4l2-python3'. No files were found to uninstall.
Successfully installed v4l2-python3-0.3.4
```

sassanh, 2024-03-29:
We are already installing it: https://github.com/ubopod/ubo-app/actions/runs/8460684596/job/23179353095#step:9:1140

It is even trying to uninstall it as seen in your command output you posted above but fails since it has been installed in operating system:
> Attempting uninstall: v4l2-python3
>     Found existing installation: v4l2-python3 0.3.2
>     Not uninstalling v4l2-python3 at /usr/lib/python3/dist-packages, outside environment /opt/ubo/env
>     Can't uninstall 'v4l2-python3'. No files were found to uninstall.

I need to investigate it.

sassanh, 2024-03-30:
Camera is working for me in the latest lite image. Please reopen this issue if it happens on other images, also mention the image.

ubopod, 2024-03-30:
@sassanh This i

ubopod, 2024-03-30:
@sassanh I realized that I am trying to install `v4l2-python3` with `pip` into `venv` but previously the package was installed system-wide with `apt install python3-v412` which has installed version `0.3.2`. When install with pip into virtual env, it installed version `0.3.4` and uses that version from `vend` instead of system-wide installed version.

ubopod, 2024-03-30:
@sassanh here is the solution:

Image release `0.12.0` is using version `0.3.2` for `python3-v412` which currently breaks:

```
_tmp_0@raspberrypi:~ $ pip show v4l2-python3
Name: v4l2-python3
Version: 0.3.2
Summary: Python bindings for the v4l2 userspace api.
Home-page: https://pypi.org/project/v4l2-python3/
Author: 
Author-email: 
License: GPLv2
Location: /usr/lib/python3/dist-packages
Requires: 
Required-by: 
```

I reinstalled the package:

```
sudo apt reinstall python3-v412
```

This bumps the version to `0.3.3` 

```
_tmp_0@raspberrypi:~ $ pip show v4l2-python3
Name: v4l2-python3
Version: 0.3.3
Summary: Python bindings for the v4l2 userspace api.
Home-page: https://pypi.org/project/v4l2-python3/
Author: 
Author-email: 
License: GPLv2
Location: /usr/lib/python3/dist-packages
Requires: 
Required-by: 
```

after a restart the camera started working again.

It seems like the debian `python3-v412` is slightly behind the `pip` package version.

sassanh, 2024-04-15:
@mehrdadfeller can you confirm this issue is resolved?

closed 2024-04-30 by sassanh