ubopod/ubo_app #57: Add WiFi error after scanning QR code

mehrdadfeller, 2024-03-29:
After scanning the QR code I get the following error in the logs. The screen get stuck on `creating scanned wifi connection` message.

```
1711697044.134122 [ERROR] 'NetworkDeviceWireless' object has no attribute '_remote_object_path'
Traceback (most recent call last):
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/utils/async_.py", line 29, in wrapper
    await awaitable
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/030-wifi/pages/create_wireless_connection.py", line 70, in act
    await add_wireless_connection(
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/030-wifi/wifi_manager.py", line 282, in add_wireless_connection
    wifi_device._remote_object_path,  # noqa: SLF001
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NetworkDeviceWireless' object has no attribute '_remote_object_path' - extra: {
    "awaitable": "<coroutine object CreateWirelessConnectionPage.create_wireless_connection.<locals>.act at 0x7f2c225930>"
```

mehrdadfeller, 2024-03-29:
@sassanh Is it possible to collect such errors via Sentry?

sassanh, 2024-03-30:
> @sassanh Is it possible to collect such errors via Sentry?

@mehrdadfeller Yes, unfortunately, the SENTRY_DSN didn't make it into the production builds in the previous images, I'm debugging the issue.

sassanh, 2024-03-30:
It was because of this commit: https://github.com/python-sdbus/python-sdbus/commit/1066153f272fa5b58314dc3cfbcce39e7c7e35ce
The latest version of sdbus released few days ago had this change, they move `_remote_object_path` and all private attributes into `_dbus` and `_dbus_meta`, in our case we are interested in `_dbus.object_path`.
I updated our code to work with their latest changes and tested it, it is working.

sassanh, 2024-03-30:
Also sentry is now working and reporting exceptions.

closed 2024-03-30 by sassanh

igo95862, 2024-05-03:
> It was because of this commit: [python-sdbus/python-sdbus@1066153](https://github.com/python-sdbus/python-sdbus/commit/1066153f272fa5b58314dc3cfbcce39e7c7e35ce) The latest version of sdbus released few days ago had this change, they move `_remote_object_path` and all private attributes into `_dbus` and `_dbus_meta`, in our case we are interested in `_dbus.object_path`. I updated our code to work with their latest changes and tested it, it is working.

Hello, python-sdbus author here.

Sorry you had to adjust your code but undocumented API is a subject to breaking changes. If there some kind of functionality you want to use from the undocumented API please open an issue so a public API can be discussed and added.

zytegalaxy, 2024-05-03:
@igo95862 thank you for developing `python-sdbus` and making it available to the developer community. We totally understand that such breaking changes can happen and fixing it just required a very minor change.