ubopod/ubo_app #58: Raspberry Pi sound card is not getting blacklisted

ubopod, 2024-03-30:
Image version: 
[ubo_app-0.12.0-bookworm.img.gz](https://github.com/ubopod/ubo-app/releases/download/0.12.0/ubo_app-0.12.0-bookworm.img.gz)

despite the file following file existing, the system is still loading the kernel module:

`/etc/modprobe.d/raspi-blacklist.conf`

```
root@raspberrypi:/opt/ubo/env/lib/python3.11/site-packages/ubo_app/system# lsmod | grep snd_bcm2835
snd_bcm2835            28672  2
snd_pcm               139264  8 snd_soc_bcm2835_i2s,snd_bcm2835,snd_soc_hdmi_codec,snd_compress,snd_soc_simple_card_utils,snd_soc_core,snd_soc_wm8960,snd_pcm_dmaengine
snd                   110592  21 snd_seq,snd_seq_device,snd_bcm2835,snd_soc_hdmi_codec,snd_timer,snd_compress,snd_soc_core,snd_pcm,snd_soc_wm8960
```

The audio device also shows up on `aplay`:

```
# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 3: wm8960soundcard [wm8960-soundcard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```

This is interfering with LED ring operation.

mehrdadfeller, 2024-03-30:
@sassanh It seems like the blacklisting takes place after a reboot:

```
root@raspberrypi:/home/_tmp_0# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: wm8960soundcard [wm8960-soundcard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
  ```

mehrdadfeller, 2024-03-31:
@sassanh This issue is a by-product of `/boot/firmware/config.txt` not being found in the first boot. Apparently that causes issues with kernel module loading process. This issue does NOT appear in `ubo_app-0.12.0-bookworm-lite.img.gz` version

sassanh, 2024-03-31:
@mehrdadfeller I think this issue and other issues related to the build process, the ones that worked alright in lite version but were broken in the default image, will be fixed in the 0.12.1, it should be released in ~2hours hopefully 🤞
Please check it once it is released and let me know if any of these issues are fixed. I keep them for you to close.