ubopod/ubo_app #63: ollama docker installation fails

mehrdadfeller, 2024-04-12:
```
  File "/opt/ubo/env/lib/python3.11/site-packages/docker/errors.py", line 39, in create_api_error_from_http_exception
    raise cls(e, response=response, explanation=explanation) from e
docker.errors.ImageNotFound: 404 Client Error for http+docker://localhost/v1.45/images/ollama/ollama:latest/json: Not Found ("No such image: ollama/ollama:latest") - extra: {
    "image": "ollama",
    "path": "ollama/ollama:latest"
```

sassanh, 2024-04-14:
@mehrdadfeller It should be fixed in 0.12.5, it is being built

ubopod, 2024-04-15:
@sassanh I was able to fetch and start `ollama` docker image on release version 0.12.5. After doing that I was also able to fetch ollama webUI. But Ollama web UI does not start. I see the following error in the log after I press start. I am not sure if that is relevant though...

```
1713152095.602765 [ERROR] Task failed
Traceback (most recent call last):
  File "/opt/ubo/env/lib/python3.11/site-packages/docker/api/client.py", line 265, in _raise_for_status
    response.raise_for_status()
  File "/usr/lib/python3/dist-packages/requests/models.py", line 1021, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 400 Client Error: Bad Request for url: http+docker://localhost/v1.45/containers/create

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/utils/async_.py", line 44, in wrapper
    await awaitable
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/080-docker/image.py", line 338, in act
    docker_client.containers.run(
  File "/opt/ubo/env/lib/python3.11/site-packages/docker/models/containers.py", line 873, in run
    container = self.create(image=image, command=command,
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ubo/env/lib/python3.11/site-packages/docker/models/containers.py", line 932, in create
    resp = self.client.api.create_container(**create_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ubo/env/lib/python3.11/site-packages/docker/api/container.py", line 439, in create_container
    return self.create_container_from_config(config, name, platform)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ubo/env/lib/python3.11/site-packages/docker/api/container.py", line 456, in create_container_from_config
    return self._result(res, True)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ubo/env/lib/python3.11/site-packages/docker/api/client.py", line 271, in _result
    self._raise_for_status(response)
  File "/opt/ubo/env/lib/python3.11/site-packages/docker/api/client.py", line 267, in _raise_for_status
    raise create_api_error_from_http_exception(e) from e
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ubo/env/lib/python3.11/site-packages/docker/errors.py", line 39, in create_api_error_from_http_exception
    raise cls(e, response=response, explanation=explanation) from e
docker.errors.APIError: 400 Client Error for http+docker://localhost/v1.45/containers/create: Bad Request ("conflicting options: hostname and the network mode") - extra: {
    "awaitable": "<coroutine object _run_container_generator.<locals>.run_container.<locals>.act at 0x7f4c68e820>"
}
```

ubopod, 2024-04-15:
I tried this on lite image. Might be we are on different images? I can fetch the image but it won't start...I fetched Ollama first and started it, I then fetched Ollama web UI but it won't start...

sassanh, 2024-04-15:
@mehrdadfeller are you seeing the same error in the logs?

mehrdadfeller, 2024-04-16:
this is working on the latest lite image (0.12.6)

closed 2024-04-16 by mehrdadfeller