ubopod/ubo_app #66: Persist application's setup/config information

mehrdadfeller, 2024-04-20:
We need a way to persist application setup/config information such as access keys, user specific configuration, as well as application data. This is a large task and we can potentially break it into smaller ones. 

It should be possible also for the user to recover their system if for example MicroSD card burns; we either need a cloud-based storage or secondary storage (such as EEPROM or EMMc). 

EEPROM is ideal for storing small amount of data (I believe currently its max size is 64KB but we can use a larger memory such as 128KB if necessary down the road). This would be great for config data or persisting access tokens. 

We need to consider security implications and potentially perform encryption at rest.

sassanh, 2024-04-20:
@mehrdadfeller We can use hashicorp vault for storing security-sensitive data, it has a docker image too: https://hub.docker.com/r/hashicorp/vault

We can implement 3 layers of security:
1. Minimum security: secrets are saved as raw data in the home directory, we just set the Linux file permissions and protect the file with SELinux.
2. A little more security: if user installs hashicorp vault docker image, we run it in dev mode and store secrets in it.
3. High security: We let user configure the vault in production mode and show them the QRCode of unseal key and initial root token, they store these keys either as qrcode or text in a secure storage (like iPhone's vault, etc). Then they need to unseal the vault either in the webui of the docker or by directly entering the key as qrcode.

For user configuration we can store them in the home directory (`XDG_CONFIG_HOME`)

We can provide configuration options for enabling back up for their config and/or secrets in the EEPROM or a secondary storage of their choice, we can add support for cloud solutions later.

sassanh, 2024-04-21:
fsspec provides an abstraction layer for file system and supports ssh-fs, Google Drive, S3, Google Cloud, Azure, Alibaba Cloud. We better use its interface since the beginning to make it easier to add cloud support in the future.