ubopod/ubo_app #70: SSH improvements

mehrdadfeller, 2024-04-21:
Some suggestions:

- Show Ip address on SSH page in addition to username and password
- Set hostname to something unique (ubo-xy.local); this way users can just remember their device hostname even if IP changes
- Show hostname on ssh
- Sometimes l and I (capital i and lower case l cannot be distinguished in password) - perhaps this character can be avoided when generating the password. This can happen sometimes 0 and O. examples:
    - BalVRFQS -> lowercase L
    - BaIVRFQS -> capital i

mehrdadfeller, 2024-04-23:
Some suggestion on updating the look / info on page (ignore the ssh icon, just a placeholder):

[image]

sassanh, 2024-04-28:
I didn't add IP, as the device may have multiple IPs in different networks. I considered showing the IPs comma separated but it would make things crowded.
You can check it in version 0.13.2

closed 2024-04-28 by sassanh