ubopod/ubo_app #79: VS code CLI download seems to get stuck v 0.13.5

zytegalaxy, 2024-05-03:
The screen gets stuck as checking... on release version `v0.13.5`

Similar to docker pull and ubp_app update progress bar discussion, adding a progress bar to indicate download status can drastically improve user experience as well.

[image]

zytegalaxy, 2024-05-03:
@sassanh we can add the progress bar to this later. First attempt should be made to make it functional. We will revisit this later for progress bar and other UX improvements

sassanh, 2024-05-08:
Please check it in version 0.14.0, it should get released in 30 minutes

closed 2024-05-08 by sassanh