ubopod/ubo_app #83: First time boot audio issue v0.13.5

zytegalaxy, 2024-05-05:
I wanted to report a strange behavior I observed during first time boot of version `0.13.5`

The audio was not working and I got the following errors after sshing and testing the audio. I then decided to restarted and audio started working after restart:

```sh
ubo@ubo-qp:/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/000-sound/sounds $ aplay failure.wav 
ALSA lib pcm_dmix.c:999:(snd_pcm_dmix_open) unable to open slave
aplay: main:831: audio open error: Invalid argument
```

```sh
ubo@ubo-qp:/home/_tmp_0 $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: wm8960soundcard [wm8960-soundcard], device 0: fe203000.i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

zytegalaxy, 2024-05-06:
@sassanh I made wrote the same image over another MicroSD card and this time, the audio worked after the first time boot. I suspect this might be a race condition or something of that nature can sometimes cause an issue...

sassanh, 2024-05-06:
@zytegalaxy Thanks! I have a couple of guesses for the reason it behaves unstable, I will check them when I come to this issue.
The thing is I really doubt if the issue is in our setup process or the first boot. So the issue should show itself for a normal device randomly after some restarts, otherwise we may consider hardware issues too.

mehrdadfeller, 2024-05-06:
Okay, I will keep an eye on this and see if it reappears. It could indeed
be a finicky hardware issue especially since this is the first time setup a
new hardware device (I have three live devices so far). The hardware
factory label tests have passed though.

Let's keep this issue open for now and I will also investigate possible
hardware causes.

zytegalaxy, 2024-05-12:
@sassanh I saw this again yesterday on v0.14.2-lite a different hardware (this is the hardware I have been using for a long time and have not witnessed any hardware related issue on it so far). The problem went away after a restart. I think this is probably some sort of race condition either during build process or boot-up. 

By the way, I have never seen this problem re-emerge after the restart. In other words, it always appears (IF appears) on the first boot-up.

sassanh, 2024-05-12:
@zytegalaxy 
It can't be an issue in the build process, because you tried the same image again and it worked.
Let's collect all logs of the system next time this issue happens.

mehrdadfeller, 2024-06-14:
log is throwing a lot of `[ERROR] No audio card found` errors:

```shell
ubo@ubo-hn:~ $ tail -f /opt/ubo/ubo-app.log
StopIteration
1718336056.297971 [ERROR] No audio card found
Traceback (most recent call last):
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/000-sound/audio_manager.py", line 59, in initialize_audio
    next(card for card in cards if 'wm8960' in card),
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
StopIteration
1718336056.300226 [ERROR] No audio card found
Traceback (most recent call last):
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/000-sound/audio_manager.py", line 59, in initialize_audio
    next(card for card in cards if 'wm8960' in card),
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
StopIteration
1718336056.301678 [ERROR] No audio card found
Traceback (most recent call last):
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/000-sound/audio_manager.py", line 59, in initialize_audio
    next(card for card in cards if 'wm8960' in card),
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
StopIteration
1718336056.303279 [ERROR] No audio card found
Traceback (most recent call last):
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/000-sound/audio_manager.py", line 59, in initialize_audio
    next(card for card in cards if 'wm8960' in card),
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
StopIteration
```
`aplay` is not listing the WM8960 as sound card.

```shell
ubo@ubo-hn:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
ubo@ubo-hn:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: 10 -- -- -- -- -- -- -- -- -- 1a -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- 58 -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
```
`1a` means the driver is not loaded since if it is loaded, it would appear as `UU`.

In the `dmesg` log below `[   12.603707] wm8960 1-001a: Failed to issue reset` error appears which means the audio chip failed to reset. I wonder if the reset message goes through I2C bus and there's some write-collision on the bus causing this.

```shell
ubo@ubo-hn:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.6.28+rpt-rpi-v8 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.5
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000e400000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000000e400000..0x000000002e3fffff (524288 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x000000003ef64520..0x000000003ef6491f (1 KiB) nomap non-reusable nvram@1
[    0.000000] OF: reserved mem: 0x000000003ef64960..0x000000003ef64995 (0 KiB) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[... 447 lines cut ...]
[   26.832022] Bluetooth: BNEP filters: protocol multicast
[   26.832046] Bluetooth: BNEP socket layer initialized
[   26.849220] Bluetooth: MGMT ver 1.22
[   26.877258] NET: Registered PF_ALG protocol family
[   27.048313] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2c@7e804000/status
[   28.337106] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2s@7e203000/status
[   30.175126] platform soc:sound: deferred probe pending
[   31.243198] systemd[785]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set
[   31.347304] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[   31.349812] bcmgenet fd580000.ethernet eth0: Link is Down
[   31.381501] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   32.129667] systemd-journald[235]: File /var/log/journal/1cac1cb88e0542189554c33d33dad653/user-1001.journal corrupted or uncleanly shut down, renaming and replacing.
[   50.664708] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[   50.679410] Bridge firewalling registered
[   51.420606] Initializing XFRM netlink socket
[   52.671849] docker0: port 1(veth8e51573) entered blocking state
[   52.671873] docker0: port 1(veth8e51573) entered disabled state
[   52.671916] veth8e51573: entered allmulticast mode
[   52.672079] veth8e51573: entered promiscuous mode
[   53.879901] eth0: renamed from veth0147243
[   53.905929] docker0: port 1(veth8e51573) entered blocking state
[   53.905949] docker0: port 1(veth8e51573) entered forwarding state
```

[image]

mehrdadfeller, 2024-07-20:
I believe I have a fix for this issue. 

I think the issue is related to reset signal not being sent to the Audio chip via i2c bus during driver initialization / loading.

In the previously posted dmesg log, the line  `[   12.603707] wm8960 1-001a: Failed to issue reset` error appears which indicated the audio chip failed to reset. Looking further in the wm8960 driver code below, it appears that this function is called inside `wm8960_i2c_probe` which is called during driver binding. 

https://github.com/waveshare/WM8960-Audio-HAT/blob/259d282dc4856318c6961e7d6a94b761308653d3/wm8960.c#L1470

This could be due to some underlying problem such as I2C bus not being ready when sending the reset command due to either being bring busy or not initialized yet during boot-up. 

Unbinding and binding the driver by the following commands seemed to fixed the issue:

```sh
echo -n "1-001a" | sudo tee /sys/bus/i2c/drivers/wm8960/unbind
echo -n "1-001a" | sudo tee /sys/bus/i2c/drivers/wm8960/bind
```

where `1-001a` means bus `1` and and device address `0x1a`. Which is the case here. 

To confirm unbind, you can call `i2cdetect 1 y` and see device address `1a` in the address table after unbind; if driver is bound, it appears as `UU` on that address after bind.

One possible solution is to use the script below and run it as service during boot-up. 
 
```sh
#!/bin/bash

# Path to the device and driver
DEVICE="1-001a"
DRIVER_PATH="/sys/bus/i2c/drivers/wm8960"

# Unbind the device
echo -n "$DEVICE" | sudo tee $DRIVER_PATH/unbind

# Short sleep to ensure the operation completes
sleep 1

# Bind the device
echo -n "$DEVICE" | sudo tee $DRIVER_PATH/bind

echo "Device has been rebound."
```

We need to ensure the system is fully booted and all other services loaded before running the rebind service.

```sh
[Unit]
Description=Rebind WM8960 driver
After=multi-user.target  # Ensures the system is fully booted and services loaded

[Service]
Type=oneshot
ExecStart=/usr/local/bin/rebind_wm8960.sh

[Install]
WantedBy=multi-user.target
```

sassanh, 2024-08-01:
@mehrdadfeller 
It just happened to me that the device after its initial boot didn't have audio and I ran the unbind and bind commands and it started working, as you suggested earlier it should be a race condition.
So I'm going to run these two commands whenever the audio device generates this particular error I just got here, but I mark it as a temporary patch in our codebase so that after we resolve the issue upstream we remove it.

sassanh, 2024-08-01:
aplay error:

```
ALSA lib pcm_dmix.c:999:(snd_pcm_dmix_open) unable to open slave
aplay: main:831: audio open error: Invalid argument
```

Exception caught in our app:

```
_simpleaudio.SimpleaudioError: Error opening PCM device. -- CODE: -22 -- MSG: Invalid argument
```

After running the command, aplay started working and ubo-app also started working without even needing to restart its process.

commit 46f1ca46 (2024-08-01): fix(audio): add a recovery mechanism for audio service to rebind the sound card if it is not available - closes #83
commit f5ee50e9 (2024-08-01): fix(audio): add a recovery mechanism for audio service to rebind the sound card if it is not available - closes #83
closed 2024-08-01 by sassanh