ubopod/ubo_app #85: Update audio instructions when camera button is presses during WiFi onboarding

zytegalaxy, 2024-05-06:
Please use picovoice orca (TTS) engine with female voice to **pre-record** the following message and play whenever the user presses the camera button during WiFi onboarding process. The existing audio is my voice and not consistent with other voice prompts.

`Show QR code to the camera to connect to WiFi`

You might have to use phonetic symbols to get the engine pronounce QR code and WiFi correctly.

sassanh, 2024-05-06:
@zytegalaxy Sure, please be aware that currently we are generating all voices in runtime, to prerecord audio files in an automated and sophisticated way, we need to implement a mechanism like [Babel](https://babel.pocoo.org/en/latest/) or even babel itself.
Such solution would search our codebase for all strings with a certain pattern (for example `voice('some string')`) and automatically generate voice files for them and distribute them in the python packages (but not in the git repository), it happens in GitHub actions. Then in runtime whenever such voice is reached, it will automatically load the file from the directory of voice files or otherwise it will try to generate it in runtime.

mehrdadfeller, 2024-05-06:
I think for this very specific issue, we can just run picovoice TTS once and output the text to a file and change the existing instruction file with the new one. 

I am for the suggestion to abstract the TTS and STT calls into `say('say something')` or `listen()`  and search in a local directory if speech for given string has been generated before. A simple solution would be to use the exact string or hash of it as the audio filename. 

We can discuss the generic solution for this and plan for it in the meantime.

closed 2024-08-17 by sassanh