ubopod/ubo_app #90: Issue with getting Ngrok up and running

mehrdadfeller, 2024-05-08:
I tried to get Ngrok running but ran into several issues; It was not clear what needed to get converted to QR code (`authtoken` or the full url mentioned in the info section). I tried both; `authtoken` gave me the first error below and full URL was accepted but the displayed showed the 3rd picture below. 

I believe we should validate the string format in the qrcode and show error message such as `invalid input` given the format of the token for each service...

[image]
[image]
[image]

sassanh, 2024-05-08:
> It was not clear what needed to get converted to QR code (authtoken or the full url mentioned in the info section).

The authtoken should be converted to QR code, it’s not:
```
Get the token from ngrok dashboard in the format “https://dashboard.ngrok.com/auth/yourtoken”
```

that url is the place you can get your token from, maybe we need to put a colon between “dashboard” and the url or put the url in parentheses.

If it is not accepting the token, we need to check the logs, no special format is required for the token, it can be any string, unfortunately the ngrok token has no known format, I have even seen tokens with hugely different lengths. It’s mostly following a format but not always. I think we can instead invest on improving the prompt message and the final error message.

Btw entering access token didn’t gave you any errors, please note that the second screen is not an error, it’s not red or doesn’t have any error icon or any mention of “failure” in its message, it is the next input it needs in order to run the ngrok docker container. It is needed to determine the port that is going to be tunneled. Entering that url instead of a valid ngrok command has caused the container to fail.

zytegalaxy, 2024-05-08:
got it I got confused by the UX. It seems like I did it correctly the first time and the `authtoken` was accepted. I didn't realize I have to make a second QR code to enter service port. Maybe we can improve this UX. Some possibilities come to my mind:

1. when we are generating the QR code, we generate it explicitly for Ngrok and ask for all necessary info, and embed it into the QR code. 

2. Also maybe we can have a sub-menu to select `tcp`, `http` options and let the user use the up/down buttons to set the port number. We need to implement a fast scroll (hold down) to increment/decrement port numbers.

3. maybe we can build an experience similar to setting the time on a watch using the side buttons (one button change the digit and other buttons increment/decrement the digit). In this case we can have 5 digits (65535), and use can select each digit and set it. 

4. Another possibility is to allow user select common ports  (22, 80, 8080, etc) for a list of options...

zytegalaxy, 2024-05-08:
btw, I don't think we need to show the URL where to get the token from... it is quite obvious and also the URL can change at any time. We would probably say something like "Log into your Ngrok account and get the authtoken from your dashboard and convert it to QR code"

closed 2024-05-10 by sassanh