ubopod/ubo_app #114: VSCode Remote Tunnel background system service requires restart after login/logout

mehrdadfeller, 2024-05-15:
It appears that if the user has installed VSCode Remote Tunnel background system service and logs out, service still keeps the user login until it is restarted. The same issue applies to login (service won't detect a login until a restart).

To address this, I suggest adding a `Service` or `Manage Service` menu item that when pressed presents the user with following options:

- Install/Uninstall Service
- Restart 
- Stop
- Start (might not be needed...)

This way the user can explicitly restart the service manually with this option after logging in and logging out.

We need to also make it clear in the UI that a service restart is required for login and logout to take effect after the service is installed and running.

mehrdadfeller, 2024-09-14:
to reproduce this issue log in once, then install service, then logout and try to login again and you won't be able to connect to tunnel.

sassanh, 2024-09-26:
Done in b79e84dc319d9027aa840a0baf481169bbf50a2b

commit cc1d1356 (2024-09-26): fix(vscode): stop and uninstall vscode service when logged out and install and start it when logged in - fixes #114
commit 3ace33c0 (2024-09-27): fix(vscode): stop and uninstall vscode service when logged out and install and start it when logged in - fixes #114
commit 394157c4 (2024-09-27): fix(vscode): stop and uninstall vscode service when logged out and install and start it when logged in - fixes #114
closed 2024-09-26 by sassanh