ubopod/ubo_app #116: Force restart Ubo app with rear-panel push button

mehrdadfeller, 2024-05-16:
We might have to change this issue title and the way to implement this. 

Currently if the Ubo app freezes (which shouldn't in the first place but can happen), the only resource is to completely restart the system. We could offer a way to kill and restart the app but we can't use the keypad buttons for this. I have added a push button at the back of device. The design intention was to use that as a shortcut for device restart; but we can probably use that to initially just restart the app via a side channel service. 

Just a high level UX idea and we can talk about / plans to implement it.

sassanh, 2024-05-29:
@mehrdadfeller Is it connected to a GPIO pin?

mehrdadfeller, 2024-06-16:
The rear push button is connected to GPIO27. Also, please ensure that when device is booted up GPIO17 is pulled down otherwise pressing the button can interfere with I2C bus and makes it fail. Pulling down GPIO17 isolates the bus from the push button signal.

[image]

mehrdadfeller, 2024-07-30:
@sassanh This should be restarting Ubo app rather than restarting the whole OS. Maybe we can change this so that a short press restarts the app and a long press restarts the OS?

commit dc5ae3a4 (2024-07-20): feat(core): long-pressing the reset button for 3 seconds or more reboots the device - closes #116
commit 2eb00ab8 (2024-07-20): feat(core): long-pressing the reset button for 3 seconds or more reboots the device - closes #116
commit 27cebb7f (2024-08-03): refactor(core): kill the app (so that it restarts) when the reset button is pressed for 0.5 seconds or more and reboot the device when it is pressed for 3 seconds or more - closes #116
closed 2024-08-03 by sassanh