ubopod/ubo_app #126: App freezes after 30~40 hours of activity

zytegalaxy, 2024-05-29:
This is a known issue but I didn't have an issue for it here. We need to increase the priority for this since it can damage the user experience.

sassanh, 2024-05-29:
@zytegalaxy 
I think vscode cli is the root of this issue.
Since you have many devices, can you add `UBO_DISABLED_SERVICES=vscode` environment variable to the service file, reload local systemctl with `systemctl --user daemon-reload` and then restart the ubo-app service and see if it happens again or not?
Please make sure vscode is disabled, its menu item should be removed.

nubcore, 2024-06-26:
reproduced issue with v0.15.0 build freshly installed. attempted work around steps noted edited `/etc/systemd/system/ubo-system.service` to add param. System didn't do anything with `--user` param but ran it without as well and then it restarted
`/etc/systemd/system $ sudo systemctl daemon-reload`
`/etc/systemd/system $ sudo systemctl restart ubo-system.service `

Ring LEDs flashed but screen UI still locked.

mehrdadfeller, 2024-07-13:
@nubcore thanks for the feedback. we are still investigating this one. Please let me know if you have any new findings.

sassanh, 2024-07-19:
# Legend

- [x] Successful
- [ ] Failed

# First round:
- [x] UBO_DISABLED_SERVICES=sound,notifications,voice,keyboard,keypad,ethernet,ip
- [x] UBO_DISABLED_SERVICES=notifications,keyboard,ethernet,wifi,rgb_ring,lightdm,vscode
- [ ] UBO_DISABLED_SERVICES=sound,keyboard,keypad,wifi,camera,lightdm,ssh **Failure time not recorded**
- [x] UBO_DISABLED_SERVICES=sound,notifications,voice,wifi,camera,rgb_ring,sensors

# Second round:
- [x] UBO_DISABLED_SERVICES= everything except sound was disabled
- [x] UBO_DISABLED_SERVICES= everything except sound was disabled
- [x] UBO_DISABLED_SERVICES= everything except sound and sensors was disabled
- [x] UBO_DISABLED_SERVICES= everything except sound and sensors was disabled
- [ ] UBO_DISABLED_SERVICES= sound and sensors **Failure time not recorded, we think it was less than 15 hours**

# Third round:
- [x] UBO_DISABLED_SERVICES=keyboard,keypad,ethernet,ip,wifi,camera,rgb_ring,sensors,lightdm,ssh,vscode,docker
- [x] UBO_DISABLED_SERVICES=sound,notifications,voice,ip,wifi,camera,rgb_ring,sensors,lightdm,ssh,vscode,docker
- [x] UBO_DISABLED_SERVICES=sound,notifications,voice,keyboard,keypad,ethernet,rgb_ring,sensors,lightdm,ssh,vscode,docker
- [x] UBO_DISABLED_SERVICES=sound,notifications,voice,keyboard,keypad,ethernet,ip,wifi,camera,ssh,vscode,docker
- [x] UBO_DISABLED_SERVICES=sound,notifications,voice,keyboard,keypad,ethernet,ip,wifi,camera,rgb_ring,sensors,lightdm

# Fourth round:
- [ ] UBO_DISABLED_SERVICES=keyboard,keypad,ethernet,ip,wifi,camera,rgb_ring,sensors,lightdm,ssh **Failure in around ~24 hours**
- [x] UBO_DISABLED_SERVICES=sound,ip,wifi,camera,rgb_ring,sensors,lightdm,ssh,vscode,docker
- [x] UBO_DISABLED_SERVICES=sound,notifications,voice,keyboard,rgb_ring,sensors,lightdm,ssh,vscode,docker
- [x] UBO_DISABLED_SERVICES=sound,notifications,voice,keyboard,keypad,ethernet,ip,ssh,vscode,docker
- [x] UBO_DISABLED_SERVICES=sound,notifications,voice,keyboard,keypad,ethernet,ip,wifi,camera,rgb_ring

* Not sure if second and third devices failed or not as they were turned off during the experiment and we didn't have access to their logs.

# Fifth round (running, no crash after 30 hours):
- [ ] UBO_DISABLED_SERVICES=notifications,voice,keyboard,keypad,ethernet,ip,wifi,camera,rgb_ring,sensors,lightdm,ssh,vscode
- [ ] UBO_DISABLED_SERVICES=notifications,voice,keyboard,keypad,ethernet,ip,wifi,camera,rgb_ring,sensors,lightdm,ssh,docker

commit 0bd9137d (2024-07-29): fix(ip): close sockets opened for testing the internet connection - closes #126
closed 2024-07-29 by sassanh