ubopod/ubo_app #153: The UX of the update process is misleading

sassanh, 2024-08-04:
After the user initiates the update process, ubo-app starts downloading the wheels, when the download is done, it reboots the system and as system boots, update-manager service runs and tries to install the downloaded wheels, the problem is at the same time ubo-app runs without any indicator to show the user an update is in progress, so when the user goes to about and checks the version, they see their system is not updated after reboot and may be confused. The actual update happens when update-manager silently installs the wheels in the background and initiates another reboot and after this second reboot the new version runs.

We need to change this process:
1. We need to inform the user about the update plan in the first place and tell them the update will be complete after the second reboot.
2. We need to make sure the user cannot interact with the update cache while update-manager is installing packages in the background, we can reuse the update-manager filesystem lock for this purpose, we can even consider dropping the write permissions in the filesystem layer when the update is in progress.
3. We need an indicator in the ubo-app (preferably a progress bar), informing the user about update-manager installing wheels in the background.
4. We can add a welcome screen (a simple notification) informing user their update was completed successfully after the second reboot.

sassanh, 2024-08-10:
@mehrdadfeller If I'm not mistaken, you wanted to design this in Figma before implementation so I assigned it to you.
Please reassign it to me if you think a Figma design is not needed and the four changes mentioned above are enough.

mehrdadfeller, 2024-08-12:
@sassanh we don't need a separate figma design for this. When the user presses update we should show create active notification which shows a pop-up similar to a regular notification and can be dismissed. In the notification text we can say "starting the update process. Device will restart shortly". 

Maybe we can give the user an option to install the update at the next reboot instead of automatically restarting but prompting the user before a restart. We also show the progress wheel circle on the top header section of the screen.

After the restart, if the _update directory exists, we should show another active notification to the user saying "Update in progress. Please keep the device powered on. This may take around 20 minutes to complete."

Similarly, we should a progress wheel on the top header for the second phase of the update process. The progress value does not need to be accurate. It only needs to indicate that "some" progress is being made so user knows the process is not stalled.

sassanh, 2024-08-13:
The progress wheel for the second phase will be implemented in #159

commit c5e8ddac (2024-08-13): feat(core): improve the user experience of update-manager by making it more verbose about the current state of the update progress - closes #153
commit cc1343ef (2024-08-13): feat(core): improve the user experience of update-manager by making it more verbose about the current state of the update progress - closes #153
commit f52a4b0e (2024-08-13): feat(core): improve the user experience of update-manager by making it more verbose about the current state of the update progress - closes #153
commit fa8f386d (2024-08-14): feat(core): improve the user experience of update-manager by making it more verbose about the current state of the update progress - closes #153
closed 2024-08-13 by sassanh