ubopod/ubo_app #154: keypad stops working IF app is restarted with hardware button in the back

mehrdadfeller, 2024-08-06:
keypad stops working IF app is restarted with hardware button in the back BUT it functions well if I restart it by `killall -9 ubo` command. 

```
ubo@ubo-w5:/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/000-audio/sounds $ tail -n 100 /opt/ubo/ubo-app.log
1722981387.116208 [INFO] ----------------------Starting the app----------------------
1722981387.932159 [INFO] Checking for updates...
1722981388.685294 [ERROR] Thread exception
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/opt/ubo/env/lib/python3.11/site-packages/sentry_sdk/integrations/threading.py", line 72, in run
    reraise(*_capture_exception())
  File "/opt/ubo/env/lib/python3.11/site-packages/sentry_sdk/_compat.py", line 127, in reraise
    raise value
  File "/opt/ubo/env/lib/python3.11/site-packages/sentry_sdk/integrations/threading.py", line 70, in run
    return old_run_func(self, *a, **kw)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/load_services.py", line 289, in run
    result = cast(Callable[[], None], self.setup)()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/000-keypad/ubo_handle.py", line 13, in setup
    init_service()
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/000-keypad/setup.py", line 228, in init_service
    Keypad()
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/000-keypad/setup.py", line 65, in __init__
    self.init_i2c()
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/000-keypad/setup.py", line 110, in init_i2c
    self.aw = adafruit_aw9523.AW9523(i2c, self.bus_address)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ubo/env/lib/python3.11/site-packages/adafruit_aw9523.py", line 91, in __init__
    if self._chip_id != 0x23:
       ^^^^^^^^^^^^^
  File "/opt/ubo/env/lib/python3.11/site-packages/adafruit_register/i2c_struct.py", line 81, in __get__
    i2c.write_then_readinto(buf, buf, out_end=1, in_start=1)
  File "/opt/ubo/env/lib/python3.11/site-packages/adafruit_bus_device/i2c_device.py", line 140, in write_then_readinto
    self.i2c.writeto_then_readfrom(
  File "/opt/ubo/env/lib/python3.11/site-packages/busio.py", line 232, in writeto_then_readfrom
    return self._i2c.writeto_then_readfrom(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ubo/env/lib/python3.11/site-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 98, in writeto_then_readfrom
    readin = self._i2c_bus.read_i2c_block_data(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ubo/env/lib/python3.11/site-packages/Adafruit_PureIO/smbus.py", line 264, in read_i2c_block_data
    ioctl(self._device.fileno(), I2C_RDWR, request)
OSError: [Errno 5] Input/output error - extra: {
  "exception_thread": "<UboServiceThread id=keypad label=Keypad name=000-keypad>"
```

sassanh, 2024-08-10:
@mehrdadfeller We concluded it's probably a hardware issue on that particular pod.

sassanh, 2024-10-30:
@mehrdadfeller were you able to reproduce this on other devices?