ubopod/ubo_app #155: SSH option does not appear in Remote menu sometimes

mehrdadfeller, 2024-08-08:
I have not been able to reproduce this behaviour yet but in one instance of this issue happening I can see the following related error in the logs.

It seems like that Raspberry Pi Connect takes over port 22 when client is connected via remote shell and it could be the cause for ssh service failure. 

```
1722974374.458483 [ERROR] Task failed
Traceback (most recent call last):
  File "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/utils/async_.py", line 47, in wrapper
    await awaitable
RuntimeError: Task <Task pending name='Task-5676' coro=<create_task.<locals>.wrapper() running at /opt/ubo/env/lib/python3.11/site-packages/ubo_app/utils/async_.py:47>> got Future <_GatheringFuture pending> attached to a different loop - extra: {
  "awaitable": "<_GatheringFuture pending>",
  "task": "<Task pending name='Task-5676' coro=<create_task.<locals>.wrapper() running at /opt/ubo/env/lib/python3.11/site-packages/ubo_app/utils/async_.py:51>>",
  "thread_": "<UboServiceThread id=ssh label=SSH name=050-ssh>",
  "ubo_service_label": "SSH",
  "ubo_service_path": "/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/050-ssh"
}
```

sassanh, 2024-08-13:
@mehrdadfeller We need reproduction steps for this.

sassanh, 2024-10-30:
@mehrdadfeller I'm closing this as it has been a while and we haven't seen this issue, even if it was a bug, there is a good chance it has been resolved by now.

closed 2024-10-30 by sassanh