ubopod/ubo_app #157: If I update VSCODE CLI through VSCODE windows, VSCODE stops working

mehrdadfeller, 2024-08-11:
If I update VScode CLI remotely (based on the prompt inside VScode environment), tunnel does not reconnect.

> Solution:

It seems like Uninstalling and Installing VScode service fixes the issue. Perhaps we need to somehow force the uninstall/install behavior after update?

[image]

sassanh, 2024-08-11:
@mehrdadfeller Have you checked if stopping and then starting the service helps? It might be even that the service is stopped after the remote update and just needs to be started.

mehrdadfeller, 2024-08-11:
@sassanh On the GUI we only have login and logout option as well as install/uninstall service. we don't have start/stop options. I did not try doing that via the terminal though...If that happens again, I will make sure to try it.

The main issue is that if this happens when user does not have physical access to the device or can't ssh to it, the user may not be able to reconnect. That's why I was suggesting an automatic recovery.

sassanh, 2024-08-11:
@mehrdadfeller Automatic recovery is a good idea.

> On the GUI we only have login and logout option as well as install/uninstall service. we don't have start/stop options

You're right, we don't start/stop menu items!

Let's find out if the service becomes stopped after update. The thing is when it happens remotely, our app has no idea when it begins/ends its update, unless we periodically check it, say every minute.