ubopod/ubo_app #184: Let the user set/unset an account as sudoer in users menu

sassanh, 2024-10-17:


zytegalaxy, 2024-10-17:
@sassanh can the user still add `ubo` account to sudoers? (wouldn't that cause the same side effect as the wifi permission issue?) or does it have to be another user other than `ubo`?

sassanh, 2024-10-17:
@zytegalaxy 
It is a step towards our final design, later we will use this same code but for other users.
For now we can do any of these things:
1. Let this feature only work for users other than ubo.
2. Ignore it for now and instruct the user they should turn off sudo for ubo to make wifi work, or just completely ignore it in this transition phase.
3. Invest some time in the policy kits to improve our polkit rule so that it works nice with sudoers too.