ubopod/ubo_app #217: high memory usage due to some memory leak

mehrdadfeller, 2025-01-16:
This is most likely due to vs code or ip service. The memory usage goes up around 1% per day. After a few month, the memory goes to %100 and device crashes. This was observed on `v1.0.0`. It should be verified on latest versions.

zytegalaxy, 2025-02-17:
@sassanh I believe based on previous conversation, this was due to vscode software CLI changes. You suggested the following as a possible fix:

[image]

Also, one theory why the memory usage was increasing could be related to vscode issues dispatching too many notifications that remain open. we need to investigate further.

sassanh, 2025-02-26:
@zytegalaxy Thanks! Also related for context: https://github.com/microsoft/vscode/issues/192225#issuecomment-1731713207

commit 85d40520 (2025-02-27): fix: vscode binary now needs to be instructed about the location of the binary with `version use` sub-command - closes #217
commit 45e6b208 (2025-02-28): fix: vscode binary now needs to be instructed about the location of the binary with `version use` sub-command - closes #217
PR #279 (2025-04-04): release/v1.3.0
Change log:

- fix: remove dependencies of `publish` job of `publish_to_pypi` workflow
- fix: vscode binary now needs to be instructed about the location of the binary with `version use` sub-command - closes #217
- fix: image size being too big for the lite version of the raspberry os
- fix: restart avahi-daemon after the hostname is set in `ubo-system`
- feat: add support for `set`s and `dict`s for rpc api
- feat: `{{hostname}}`, as a template variable, will be replaced with the hostname of the device whenever used in `title` and `content` properties of a `Notification` instanc [... cut]
closed 2025-02-27 by sassanh