ubopod/ubo_app #223: running ubo-app on bare Raspberry Pi (no hardware add on) fails due to audio service existence

zytegalaxy, 2025-01-30:
I managed to address this issue but disabling the audio if there's no hardware add on. I suggest we look for audio test results in `/proc/device_tree/hat/custom_0` file (`speakers` and `microphones` fields in the JSON file) and if no audio add-on info is on EEPROM, then disable audio. This way we can store hardware enabled features in the EEPROM and put some hardware behind a feature flag.

zytegalaxy, 2025-01-31:
@sassanh I believe this is partially addressed. I think the current solution of retrying several times and aborting is good enough. I am going to lower priority of this ticket and we can revisit it later. Alternatively we can close this issue and reopen later as deemed necessary. 

Just to have a reference to offline conversation, you made a suggestion to enhance robustness of the app by making sure that each thread that managed a hardware resource (audio, camera, etc) runs independently and if one fails the app would still function.

sassanh, 2025-01-31:
@zytegalaxy Yes, I believed we had an issue for isolating threads and making the app fail-proof, but I can't find one. I will create one now. I think it has priority as it improves the stability of the app.

sassanh, 2025-02-01:
@zytegalaxy Is it already fixed with that patch we applied on webui pull request?

zytegalaxy, 2025-02-01:
I haven't tested it on bare raspberry pi but it should resolve it. I will test tomorrow

commit 98505a4c (2025-03-08): refactor(core,audio,display,keypad,sensors): read and check content of eeprom to determine if a device/service should be initialized or not - closes #223, #closes #249
commit b1b9ffce (2025-03-08): refactor(core,audio,display,keypad,sensors): read and check content of eeprom to determine if a device/service should be initialized or not - closes #223, #closes #249
commit 84b2afa1 (2025-03-08): refactor(core,audio,display,keypad,sensors): read and check content of eeprom to determine if a device/service should be initialized or not - closes #223, #closes #249
commit 02b6df82 (2025-03-09): refactor(core,audio,display,keypad,sensors): read and check content of eeprom to determine if a device/service should be initialized or not - closes #223, #closes #249