ubopod/ubo_app #224: How Ubo GUI on Web UI by default

zytegalaxy, 2025-01-30:
To close the loop and make sure folks can use the GUI without having access to hardware. Now that we have WiFi hotspot at boot-up, this feature can complete the hardware-less experience of the Ubo app. 

I believe we have the building blocks to build this but the missing part is sending the keypad events and adding the keypad buttons on the UI.

I have attached SVG design for the keypad below.

On mobile, it would be great to enlarge the keypad GUI to fix the width of the screen. 

[image]

[image]

zytegalaxy, 2025-01-30:
@sassanh btw, I am not so sure if the would be hosted on the same port 4321 or we serve this independently on a different port. It would be great if we can fuse the web UI entry with the GUI. 

For example, when you go to picovoice and want to enter credentials on the GUI, it would automatically redirect to get the credentials input and back to the GUI.

sassanh, 2025-01-30:
@zytegalaxy We can show them side by side in two columns on desktop and show inputs beneath the GUI on mobile.

The GRPC is served on a different port, 4321 is for the webui, but the web application served on 4321 will be able to connect to the GRPC port and render the GUI.

commit f0c2925f (2025-02-28): feat: implement react web application for the web-ui service - closes #224
commit c1f6a17c (2025-02-28): feat(web-ui): implement react web application for the web-ui service - closes #224
commit 85a4d792 (2025-02-28): refactor(web-ui): use mui dialogs for input demands - closes #224
commit 425adef0 (2025-02-28): feat(web-ui): implement react web application for the web-ui service - closes #224
closed 2025-03-07 by sassanh