ubopod/ubo_app #261: Bring menu object to redux

zytegalaxy, 2025-03-10:
Since redux design was introduced after UI work in `ubo_app`, it is not currently on the redux bus. It would make a lot of sense to bring design to redux as you suggested. 

This would completely decouple the UI logic from its rendering and opens a lot of new possibilities. 

It would also make debugging UI issues much easier. 

This way Web UI can be made aware of the current state of the UI and we can also communicate UI state over gRPC instead of streaming the pixel values. Other implementations of the UI can also create their own visual representation of the menu function.

zytegalaxy, 2025-03-10:
@sassanh by the way, this task could automatically revolve https://github.com/ubopod/ubo_app/issues/235 since it would move state management onto the redux store for the UI if I am not mistaken.

mehrdadfeller, 2025-05-16:
@sassanh One of the main advantages of bringing UI logic into the store is that we can complete separate the rendering and UI implementation from the UI logic. 

This opens the door to developers to implement a super thin client on MCU (Raspberry Pi Pico, ESP32, etc) that calls the gRPC API and act as low-cost & low-power detached GUI. 

For example, someone can even make a [Pebble Watch](https://ericmigi.com/blog/introducing-two-new-pebbleos-watches) GUI this way.