ubopod/ubo_app #264: SSID is not being advertised: `Hostapd` is running but `wlan0` is in managed mode

zytegalaxy, 2025-03-21:
Tested on `lite.img` file downloaded from https://github.com/ubopod/ubo_app/actions/runs/13965699252

Steps to reproduce:

1. Add WiFi
2. Delete WiFi
3. Add WiFi with Web UI over Hotspot 

SSID cannot be discovered despite `hostapd` service running.

Debug observations:

The mode reported by `iw dev wlan0 info` is `managed` instead of `ap`

Restarting `hostapd` service resolves the issue. Most likely the `wlan0` resource is not properly released to `hostapd`. We might have to consider using `nmcli` instead of `hostapd` so that the resource does not need to be handed off.

zytegalaxy, 2025-03-21:
@sassanh Can you please try to see if you can also reproduce this?

zytegalaxy, 2025-03-31:
@sassanh I tried to reproduce this after recent changes to make WiFi more stable but it did not appear. This might have been already resolved.

sassanh, 2025-03-31:
@zytegalaxy Yeah, it should be resolved, it was probably a side-effect of async tasks getting cancelled.

closed 2025-03-31 by sassanh