ubopod/ubo_app #267: sometimes `ubo-system` consumes too much CPU due to ping

zytegalaxy, 2025-03-21:
I believe sometimes `ping` goes into some bad estate (maybe the threads live on). The issue goes after after restarting 

One important observation I had was that the globe icon was red even though I could run `ping` command in shell.  This indicates that ping was not working properly. 

Next, I restarted `ubo-system` service but the globe icon remained red. However after restarting `ubo-app` the icon updated correctly.

zytegalaxy, 2025-03-27:
@sassanh Yeah this one is a bit hard to reproduce. It happens rarely and often after some time of usage. It seems like ping can get into a bad state sometimes. I will try to see if there's a way to reproduce this.

sassanh, 2025-03-27:
@zytegalaxy currently we have this code from here ubo_app/system/system_manager/main.py:63:

```python
    while True:
        try:
            response = ping('1.1.1.1', timeout=1, count=1, out=None)
            connection_state.state = (
                NetState.CONNECTED if response.success() else NetState.DISCONNECTED
            )
        except OSError:
            connection_state.state = NetState.DISCONNECTED
        time.sleep(0.25)
```

It may be that an exception other than `OSError` happens, but then it should be raised out of the while loop and crash the whole thread, at least.

Other possibility is that the client goes into a bad state and sends rapid connection queries in an unusual rate.

zytegalaxy, 2025-03-31:
@sassanh can you please connect this to relevant PR?

sassanh, 2025-03-31:
@zytegalaxy done

commit c3b456a7 (2025-03-31): refactor(ip): remove python-ping package and the connection monitoring code in the system-manager, use system ping command instead, this is because none of the python packages providing ping functionality are actively maintained and ping command has the benefit of not needing to run as root, lowering the communication traffic between the system process and the main process - closes #267
closed 2025-04-02 by sassanh