ubopod/ubo_app #281: Install hangs on 'Looking for font path...'

nubcore, 2025-04-12:
Fresh install of bookworm for Pi 4, then run ubo install

curl -sSL https://raw.githubusercontent.com/ubopod/ubo-app/main/ubo_app/system/install.sh | sudo bash

hangs trying to find fonts.

+ apt-get -y upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  chromium chromium-common chromium-sandbox firmware-libertas
  firmware-misc-nonfree labwc libcamera-ipa libcamera-tools libegl-mesa0
  libgbm1 libglapi-mesa libglx-mesa0 libpipewire-0.3-0 libpipewire-0.3-modules
  libspa-0.2-bluetooth libspa-0.2-libcamera libspa-0.2-modules
  linux-headers-rpi-2712 linux-headers-rpi-v8 linux-image-rpi-2712
  linux-image-rpi-v8 lxplug-netman piclone pipewire pipewire-bin
  pipewire-pulse pprompt python3-libcamera raindrop raspberrypi-ui-mods rc-gui
  rpicam-apps wf-panel-pi wfplug-connect wfplug-squeek
The following packages will be upgraded:
  agnostics avahi-daemon avahi-utils base-files bsdextrautils bsdutils
  cups-pk-helper curl dns-root-data dnsmasq-base eject fdisk ffmpeg firefox
  firmware-atheros firmware-brcm80211 firmware-realtek ghostscript git git-man
  gstreamer1.0-alsa gstreamer1.0-plugins-base gstreamer1.0-plugins-good
  gstreamer1.0-x gui-pkinst gui-updater initramfs-tools initramfs-tools-core
  libavahi-client3 libavahi-common-data libavahi-common3 libavahi-core7
  libavahi-glib1 libavcodec59 libavdevice59 libavfilter8 libavformat59
  libavutil57 libblkid1 libc-bin libc-dev-bin libc-devtools libc-l10n libc6
  libc6-dbg libc6-dev libcamera0.3 libcurl3-gnutls libcurl4 libebml5 libfdisk1
  libfreetype6 libgl1-mesa-dev libgl1-mesa-dri libglib2.0-0 libglib2.0-bin
  libglib2.0-data libgnutls30 libgs-common libgs10 libgs10-common
  libgstreamer-gl1.0-0 libgstreamer-plugins-base1.0-0 libgstreamer1.0-0
  libjavascriptcoregtk-4.1-0 libkpathsea6 libliftoff0 liblzma5 libmount1
  libnfsidmap1 libnm0 libnuma1 libopenh264-7 libopenjp2-7 libpam-systemd
  libpipewire-0.3-common libpisp-common libpisp1 libpixman-1-0 libpostproc56
  libpython3.11 libpython3.11-dev libpython3.11-minimal libpython3.11-stdlib
  librabbitmq4 librados2 librbd1 libsmartcols1 libsoup2.4-1 libsoup2.4-common
  libsrt1.5-gnutls libswresample4 libswscale6 libsynctex2 libsystemd-shared
  libsystemd0 libtasn1-6 libtiff6 libudev1 libuuid1 libwebkit2gtk-4.1-0
  libxslt1.1 linux-libc-dev locales lp-connection-editor lxde lxde-core
  lxpanel lxpanel-data lxplug-bluetooth lxplug-cputemp lxplug-ejecter
  lxplug-magnifier lxplug-menu lxplug-updater lxplug-volumepulse lxtask
  mesa-vdpau-drivers mesa-vulkan-drivers meson mount network-manager
  nfs-common openssh-client openssh-server openssh-sftp-server pcmanfm pipanel
  pipewire-libcamera pishutdown piwiz pixflat-icons pixflat-theme python3-av
  python3-jinja2 python3-picamera2 python3-pidng python3-pkg-resources
  python3-setuptools python3-setuptools-whl python3-simplejpeg python3-urllib3
  python3-werkzeug python3.11 python3.11-dev python3.11-minimal
  python3.11-venv raspberrypi-net-mods raspberrypi-sys-mods raspi-config
  raspi-firmware raspi-utils raspi-utils-core raspi-utils-dt
  raspi-utils-eeprom raspi-utils-otp raspinfo realvnc-vnc-server rfkill
  rp-bookshelf rp-prefapps rpi-connect rpi-eeprom rpi-imager rsync squeekboard
  ssh systemd systemd-sysv systemd-timesyncd tzdata ucf udev util-linux
  util-linux-extra vim-common vim-tiny wget xserver-common xserver-xorg-core
  xz-utils
181 upgraded, 0 newly installed, 0 to remove and 35 not upgraded.
Need to get 382 MB of archives.
After this operation, 15.8 MB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main arm64 base-files arm64 12.4+deb12u10 [70.9 kB]
Get:2 http://deb.debian.org/debian bookworm/main arm64 bsdutils arm64 1:2.38.1-5+deb12u3 [94.2 kB]
Get:3 http://archive.raspberrypi.com/debian bookworm/main arm64 libc6-dbg arm64 2.36-9+rpt2+deb12u10 [8,151 kB]
Get:4 http://deb.debian.org/debian bookworm/main arm64 libsmartcols1 arm64 2.38.1-5+deb12u3 [104 kB]
Get:5 http://deb.debian.org/debian bookworm/main arm64 util-linux-extra arm64 2.38.1-5+deb12u3 [112 kB]
Get:6 http://deb.debian.org/debian bookworm/main arm64 util-linux arm64 2.38.1-5+deb12u3 [1,143 kB]
Get:7 http://deb.debian.org/debian bookworm/main arm64 mount arm64 2.38.1-5+deb12u3 [134 kB]
Get:8 http://deb.debian.org/debian bookworm/main arm64 systemd-sysv arm64 252.36-1~deb12u1 [42.6 kB]
Get:9 http://deb.debian.org/debian bookworm/main arm64 libpam-systemd arm64 252.36-1~deb12u1 [209 kB]
Get:10 http://deb.debian.org/debian bookworm/main arm64 libblkid1 arm64 2.38.1-5+deb12u3 [143 kB]
Get:11 http://deb.debian.org/debian bookworm/main arm64 systemd-timesyncd arm64 252.36-1~deb12u1 [61.7 kB]
[... 662 lines cut ...]
Setting up python3-urllib3 (1.26.12-1+deb12u1) ...
Setting up libsoup2.4-common (2.74.3-1+deb12u1) ...
Setting up libfdisk1:arm64 (2.38.1-5+deb12u3) ...
Setting up libnuma1:arm64 (2.0.18-1~rpt1) ...
Setting up libsrt1.5-gnutls:arm64 (1.5.1-1+deb12u1) ...
Setting up mount (2.38.1-5+deb12u3) ...
Setting up libtiff6:arm64 (4.5.0-6+deb12u2) ...
Setting up libxslt1.1:arm64 (1.1.35-1+deb12u1) ...
Setting up cups-pk-helper (0.2.6-1+rpt2) ...
Setting up libgl1-mesa-dev:arm64 (24.2.8-1~bpo12+rpt2) ...
Setting up pishutdown (0.32) ...
Setting up libcurl4:arm64 (7.88.1-10+deb12u12) ...
Setting up libopenjp2-7:arm64 (2.5.0-2+deb12u1) ...
Setting up git-man (1:2.39.5-0+deb12u2) ...
Setting up lxpanel-data (0.10.1-2+rpt22) ...
Setting up libpisp1:arm64 (1.1.0-1) ...
Setting up pixflat-icons (0.27) ...
gtk-update-icon-cache: Cache file created successfully.
Setting up libpython3.11-minimal:arm64 (3.11.2-6+deb12u5) ...
Setting up curl (7.88.1-10+deb12u12) ...
Setting up libpostproc56:arm64 (8:5.1.6-0+deb12u1+rpt3) ...
Setting up realvnc-vnc-server (7.13.1.57) ...
Updating /etc/pam.d/vncserver
Updating /etc/pam.conf... done
Looking for font path...

nubcore, 2025-04-12:
After a reboot, do a update and upgrade and get this...

```
sudo apt upgrade
E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem. 
```

Running the suggested ``sudo dpkg --configure -a``` then nets the same hang....

```
Setting up python3.11-minimal (3.11.2-6+deb12u5) ...
Setting up libgs10-common (10.0.0~dfsg-11+deb12u7) ...
Setting up openssh-sftp-server (1:9.2p1-2+deb12u5) ...
Setting up gui-updater (0.6) ...
Setting up pcmanfm (1.3.2-1+rpt29) ...
Setting up libsoup2.4-1:arm64 (2.74.3-1+deb12u1) ...
Setting up openssh-server (1:9.2p1-2+deb12u5) ...
rescue-ssh.target is a disabled or a static unit not running, not starting it.
ssh.socket is a disabled or a static unit not running, not starting it.
Setting up firefox (137.0-1+rpt1) ...
Setting up libcamera0.3:arm64 (0.3.2+rpt20241119-1) ...
Setting up libpython3.11-stdlib:arm64 (3.11.2-6+deb12u5) ...
Setting up lxpanel (0.10.1-2+rpt22) ...
Setting up libavahi-common3:arm64 (0.8-10+deb12u1) ...
Setting up libglib2.0-bin (2.74.6-2+deb12u5) ...
Setting up systemd (252.36-1~deb12u1) ...
Setting up network-manager (1.42.4-1+rpt1+deb12u1) ...
Setting up vim-tiny (2:9.0.1378-2+deb12u2) ...
Setting up lxplug-menu (0.9) ...
Setting up fdisk (2.38.1-5+deb12u3) ...
Setting up nfs-common (1:2.6.2-4+deb12u1) ...
auth-rpcgss-module.service is a disabled or a static unit not running, not starting it.
nfs-idmapd.service is a disabled or a static unit not running, not starting it.
nfs-utils.service is a disabled or a static unit not running, not starting it.
proc-fs-nfsd.mount is a disabled or a static unit not running, not starting it.
rpc-gssd.service is a disabled or a static unit not running, not starting it.
rpc-statd.service is a disabled or a static unit not running, not starting it.
rpc-svcgssd.service is a disabled or a static unit not running, not starting it.
var-lib-nfs-rpc_pipefs.mount is a disabled or a static unit not running, not starting it.
Setting up lxplug-magnifier (0.16) ...
Setting up lxplug-bluetooth (0.38) ...
Setting up systemd-timesyncd (252.36-1~deb12u1) ...
Setting up git (1:2.39.5-0+deb12u2) ...
Setting up rp-bookshelf (0.26) ...
Setting up libavahi-glib1:arm64 (0.8-10+deb12u1) ...
Setting up rpi-imager (1.9.3) ...
Setting up lxplug-cputemp (0.14) ...
Setting up realvnc-vnc-server (7.13.1.57) ...
Updating /etc/pam.d/vncserver
Updating /etc/pam.conf... done
Looking for font path... 
```

nubcore, 2025-04-12:
```Ctrl+C```'d out of the hang, and seem to have gotten everything but vnc server...

```
Looking for font path... 
^C
dpkg: error processing package realvnc-vnc-server (--configure):
 installed realvnc-vnc-server package post-installation script subprocess was interrupted
Setting up xserver-common (2:21.1.7-3+rpt3+deb12u9) ...
Setting up libavcodec59:arm64 (8:5.1.6-0+deb12u1+rpt3) ...
Setting up initramfs-tools-core (0.142+rpt3+deb12u1) ...
Setting up libc-dev-bin (2.36-9+rpt2+deb12u10) ...
Setting up libswscale6:arm64 (8:5.1.6-0+deb12u1+rpt3) ...
Setting up lxtask (0.1.10-1+rpt10) ...
Setting up libavahi-core7:arm64 (0.8-10+deb12u1) ...
Setting up libavahi-client3:arm64 (0.8-10+deb12u1) ...
Setting up pipewire-libcamera (1.2.7-1~bpo12+1+rpt2) ...
Setting up python3-simplejpeg (1.8.1-1) ...
Setting up libgstreamer1.0-0:arm64 (1.22.0-2+deb12u1) ...
Setcap worked! gst-ptp-helper is not suid!
Setting up rsync (3.2.7-1+deb12u2) ...
rsync.service is a disabled or a static unit not running, not starting it.
Setting up raspi-utils-core (20250314-1) ...
Setting up lp-connection-editor (0.5) ...
Setting up libsynctex2:arm64 (2022.20220321.62855-5.1+deb12u2) ...
Setting up libc-devtools (2.36-9+rpt2+deb12u10) ...
Setting up lxde-core (11+rpt2) ...
Setting up libgs10:arm64 (10.0.0~dfsg-11+deb12u7) ...
Setting up python3.11 (3.11.2-6+deb12u5) ...
Setting up systemd-sysv (252.36-1~deb12u1) ...
Setting up initramfs-tools (0.142+rpt3+deb12u1) ...
update-initramfs: deferring update (trigger activated)
Setting up lxplug-ejecter (0.25) ...
Setting up libpython3.11:arm64 (3.11.2-6+deb12u5) ...
Setting up libavformat59:arm64 (8:5.1.6-0+deb12u1+rpt3) ...
Setting up ssh (1:9.2p1-2+deb12u5) ...
Setting up lxplug-volumepulse (0.31) ...
Setting up xserver-xorg-core (2:21.1.7-3+rpt3+deb12u9) ...
Setting up libgstreamer-plugins-base1.0-0:arm64 (1.22.0-3+rpt1+deb12u4) ...
Setting up ghostscript (10.0.0~dfsg-11+deb12u7) ...
Setting up libavfilter8:arm64 (8:5.1.6-0+deb12u1+rpt3) ...
Setting up libgstreamer-gl1.0-0:arm64 (1.22.0-3+rpt1+deb12u4) ...
Setting up lxde (11+rpt2) ...
Setting up gstreamer1.0-plugins-base:arm64 (1.22.0-3+rpt1+deb12u4) ...
Setting up python3.11-venv (3.11.2-6+deb12u5) ...
Setting up libpython3.11-dev:arm64 (3.11.2-6+deb12u5) ...
Setting up raspi-firmware (1:1.20250326-1) ...
'/boot/vmlinuz-6.6.51+rpt-rpi-2712' -> '/boot/firmware/kernel_2712.img'
'/boot/vmlinuz-6.6.51+rpt-rpi-v8' -> '/boot/firmware/kernel8.img'
Setting up libavdevice59:arm64 (8:5.1.6-0+deb12u1+rpt3) ...
Setting up avahi-daemon (0.8-10+deb12u1) ...
Setting up python3-av (12.3.0-2+rpt1) ...
Setting up libpam-systemd:arm64 (252.36-1~deb12u1) ...
Setting up libc6-dev:arm64 (2.36-9+rpt2+deb12u10) ...
Setting up ffmpeg (8:5.1.6-0+deb12u1+rpt3) ...
Setting up gstreamer1.0-x:arm64 (1.22.0-3+rpt1+deb12u4) ...
Setting up gstreamer1.0-alsa:arm64 (1.22.0-3+rpt1+deb12u4) ...
Setting up python3.11-dev (3.11.2-6+deb12u5) ...
Setting up gstreamer1.0-plugins-good:arm64 (1.22.0-5+rpt1+deb12u2) ...
Setting up avahi-utils (0.8-10+deb12u1) ...
Setting up python3-picamera2 (0.3.25-1) ...
Setting up libwebkit2gtk-4.1-0:arm64 (2.48.1-2~deb12u1) ...
Setting up raspinfo (20250314-1) ...
Setting up raspi-utils (20250314-1) ...
Setting up rpi-eeprom (27.6-1) ...
Setting up raspi-config (20250312) ...
Setting up lxplug-updater (0.23) ...
Setting up rp-prefapps (0.68) ...
Setting up raspberrypi-sys-mods (20250410) ...
Installing new version of config file /etc/udev/rules.d/99-com.rules ...
Setting up piwiz (0.89) ...
Setting up raspberrypi-net-mods (1.4.3) ...
Installing new version of config file /etc/profile.d/wifi-check.sh ...
Setting up squeekboard (1.21.0-1+rpt11) ...
Processing triggers for mailcap (3.70+nmu1) ...
Processing triggers for desktop-file-utils (0.26-1) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for gnome-menus (3.36.0-1.1) ...
Processing triggers for libc-bin (2.36-9+rpt2+deb12u10) ...
Processing triggers for man-db (2.11.2-2) ...
Processing triggers for dbus (1.14.10-1~deb12u1) ...
Processing triggers for initramfs-tools (0.142+rpt3+deb12u1) ...
update-initramfs: Generating /boot/initrd.img-6.6.51+rpt-rpi-v8
'/boot/initrd.img-6.6.51+rpt-rpi-v8' -> '/boot/firmware/initramfs8'
update-initramfs: Generating /boot/initrd.img-6.6.51+rpt-rpi-2712
'/boot/initrd.img-6.6.51+rpt-rpi-2712' -> '/boot/firmware/initramfs_2712'
Errors were encountered while processing:
 realvnc-vnc-server
```

nubcore, 2025-04-12:
Running installer again doesn't seem to go particularly well either...

```
curl -sSL https://raw.githubusercontent.com/ubopod/ubo-app/main/ubo_app/system/install.sh | sudo bash
+ set -o errexit
+ set -o pipefail
+ set -o nounset
+ USERNAME=ubo
+ UPDATE=false
+ WAIT_FOR_APP=false
+ ALPHA=false
+ WITH_DOCKER=false
+ IN_PACKER=false
+ SOURCE=ubo-app
+ export DEBIAN_FRONTEND=noninteractive
+ DEBIAN_FRONTEND=noninteractive
+ echo ----------------------------------------------
----------------------------------------------
+ echo Parameters:
Parameters:
+ echo 'UPDATE: false'
UPDATE: false
+ echo 'ALPHA: false'
ALPHA: false
+ echo 'WITH_DOCKER: false'
WITH_DOCKER: false
+ echo 'SOURCE: ubo-app'
SOURCE: ubo-app
+ echo ----------------------------------------------
----------------------------------------------
++ id -u
+ '[' 0 '!=' 0 ']'
+ '[' -z ubo ']'
+ adduser --disabled-password --gecos '' ubo
adduser: The user `ubo' already exists.
+ true
+ usermod -aG adm,audio,video,gpio,i2c,spi,kmem,render ubo
+ echo 'User ubo created successfully.'
User ubo created successfully.
++ id -u ubo
+ echo 'export XDG_RUNTIME_DIR=/run/user/1001'
+ apt-get -fy install
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libc++1-16 libc++abi1-16 libcamera0.3 libunwind-16 libwlroots12
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
+ apt-get -y update
Hit:1 http://deb.debian.org/debian bookworm InRelease
Get:2 http://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease                  
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease
Fetched 48.0 kB in 1s (49.8 kB/s)
Reading package lists... Done
+ apt-get -y upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  libc++1-16 libc++abi1-16 libcamera0.3 libunwind-16 libwlroots12
Use 'sudo apt autoremove' to remove them.
The following packages have been kept back:
  raspberrypi-ui-mods wf-panel-pi wfplug-connect wfplug-squeek
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
+ apt-get -y install accountsservice dhcpcd dnsmasq git hostapd i2c-tools libasound2-dev libcap-dev libegl1 libgl1 libmtdev1 libzbar0 python3-alsaaudio python3-apt python3-dev python3-gpiozero python3-libcamera python3-picamera2 python3-pip python3-virtualenv --no-install-recommends --no-install-suggests
Reading package lists... Done
Building dependency tree... Done
[... 408 lines cut ...]
    Found existing installation: itsdangerous 2.1.2
    Not uninstalling itsdangerous at /usr/lib/python3/dist-packages, outside environment /opt/ubo/env
    Can't uninstall 'itsdangerous'. No files were found to uninstall.
  Attempting uninstall: isort
    Found existing installation: isort 5.6.4
    Not uninstalling isort at /usr/lib/python3/dist-packages, outside environment /opt/ubo/env
    Can't uninstall 'isort'. No files were found to uninstall.
  Attempting uninstall: dill
    Found existing installation: dill 0.3.6
    Not uninstalling dill at /usr/lib/python3/dist-packages, outside environment /opt/ubo/env
    Can't uninstall 'dill'. No files were found to uninstall.
  Attempting uninstall: blinker
    Found existing installation: blinker 1.5
    Not uninstalling blinker at /usr/lib/python3/dist-packages, outside environment /opt/ubo/env
    Can't uninstall 'blinker'. No files were found to uninstall.
  Attempting uninstall: flask
    Found existing installation: Flask 2.2.2
    Not uninstalling flask at /usr/lib/python3/dist-packages, outside environment /opt/ubo/env
    Can't uninstall 'Flask'. No files were found to uninstall.
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
virtualenv 20.17.1+ds requires platformdirs<3,>=2.4, but you have platformdirs 4.3.7 which is incompatible.
Successfully installed Adafruit-Blinka-8.56.0 Adafruit-Circuitpython-ConnectionManager-3.1.3 Adafruit-PlatformDetect-3.77.0 Adafruit-PureIO-1.1.11 Kivy-Garden-0.1.5 RPi.GPIO-0.7.1 adafruit-circuitpython-aw9523-1.1.11 adafruit-circuitpython-busdevice-5.2.11 adafruit-circuitpython-neopixel-6.3.15 adafruit-circuitpython-pct2075-1.1.23 adafruit-circuitpython-pixelbuf-2.0.7 adafruit-circuitpython-register-1.10.2 adafruit-circuitpython-requests-4.1.10 adafruit-circuitpython-rgb-display-3.13 adafruit-circuitpython-typing-1.11.2 adafruit-circuitpython-veml7700-2.1.2 aiofiles-24.1.0 aiohappyeyeballs-2.6.1 aiohttp-3.11.16 aiosignal-1.3.2 betterproto-2.0.0b7 binho-host-adapter-0.1.6 black-25.1.0 blinker-1.9.0 coloredlogs-15.0.1 dill-0.3.9 docker-7.1.0 fasteners-0.19 filetype-1.2.0 flask-3.1.0 flatbuffers-20181003210633 frozenlist-1.5.0 grpclib-0.4.7 h11-0.14.0 h2-4.2.0 headless-kivy-0.12.2 hpack-4.1.0 humanfriendly-10.0 hypercorn-0.17.3 hyperframe-6.1.0 isort-5.13.2 itsdangerous-2.2.0 kivy-2.3.1 mpmath-1.3.0 multidict-6.4.3 netifaces-0.11.0 onnxruntime-1.20.1 packaging-24.2 pathspec-0.12.1 piper-phonemize-1.1.0 piper-tts-1.2.0 platformdirs-4.3.7 priority-2.0.0 propcache-0.3.1 protobuf-6.30.2 psutil-7.0.0 pulsectl-24.12.0 pvorca-1.1.1 pyftdi-0.56.0 pypng-0.20220715.0 python-dat [... cut ...] mutable-1.1.1 python-redux-0.22.2 python-strtobool-1.0.3 pyusb-1.3.1 pyzbar-0.1.9 qrcode-8.1 quart-0.20.0 rpi-ws281x-5.0.0 sdbus-0.14.0 sdbus-networkmanager-2.0.0 semver-3.0.4 sentry-sdk-2.25.1 simpleaudio-1.0.4 sympy-1.13.3 sysv-ipc-1.1.0 tenacity-9.1.2 typing-extensions-4.13.2 ubo-app-1.3.0 ubo-gui-0.13.14 werkzeug-3.1.3 wsproto-1.2.0 yarl-1.19.0
+ sed -i '/^Banner /d' /etc/ssh/sshd_config.d/rename_user.conf
sed: can't read /etc/ssh/sshd_config.d/rename_user.conf: No such file or directory
```

zytegalaxy, 2025-04-12:
Hello hello 👋!

Thanks for trying the installer and reporting the issue. I am going to try to reproduce this on my end right now and report back. While we figure out the issue, you can alternatively download the full image file from the release section.

zytegalaxy, 2025-04-12:
@nubcore I tried the installer and I am getting the sam error as your second try:

```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
virtualenv 20.17.1+ds requires platformdirs<3,>=2.4, but you have platformdirs 4.3.7 which is incompatible.
Successfully installed Adafruit-Blinka-8.56.0 Adafruit-Circuitpython-ConnectionManager-3.1.3 Adafruit-PlatformDetect-3.77.0 Adafruit-PureIO-1.1.11 Kivy-Garden-0.1.5 RPi.GPIO-0.7.1 adafruit-circuitpython-aw9523-1.1.11 adafruit-circuitpython-busdevice-5.2.11 adafruit-circuitpython-neopixel-6.3.15 adafruit-circuitpython-pct2075-1.1.23 adafruit-circuitpython-pixelbuf-2.0.7 adafruit-circuitpython-register-1.10.2 adafruit-circuitpython-requests-4.1.10 adafruit-circuitpython-rgb-display-3.13 adafruit-circuitpython-typing-1.11.2 adafruit-circuitpython-veml7700-2.1.2 aiofiles-24.1.0 aiohappyeyeballs-2.6.1 aiohttp-3.11.16 aiosignal-1.3.2 betterproto-2.0.0b7 binho-host-adapter-0.1.6 black-25.1.0 blinker-1.9.0 coloredlogs-15.0.1 dill-0.3.9 docker-7.1.0 fasteners-0.19 filetype-1.2.0 flask-3.1.0 flatbuffers-20181003210633 frozenlist-1.5.0 grpclib-0.4.7 h11-0.14.0 h2-4.2.0 headless-kivy-0.12.2 hpack-4.1.0 humanfriendly-10.0 hypercorn-0.17.3 hyperframe-6.1.0 isort-5.13.2 itsdangerous-2.2.0 kivy-2.3.1 mpmath-1.3.0 multidict-6.4.3 netifaces-0.11.0 onnxruntime-1.20.1 packaging-24.2 pathspec-0.12.1 piper-phonemize-1.1.0 piper-tts-1.2.0 platformdirs-4.3.7 priority-2.0.0 propcache-0.3.1 protobuf-6.30.2 psutil-7.0.0 pulsectl-24.12.0 pvorca-1.1.1 pyftdi-0.56.0 pypng-0.20220715.0 python-dateutil-2.9.0.post0 python-debouncer-0.1.5 python-dotenv-1.1.0 python-fake-0.2.0 python-immutable-1.1.1 python-redux-0.22.2 python-strtobool-1.0.3 pyusb-1.3.1 pyzbar-0.1.9 qrcode-8.1 quart-0.20.0 rpi-ws281x-5.0.0 sdbus-0.14.0 sdbus-networkmanager-2.0.0 semver-3.0.4 sentry-sdk-2.25.1 simpleaudio-1.0.4 sympy-1.13.3 sysv-ipc-1.1.0 tenacity-9.1.2 typing-extensions-4.13.2 ubo-app-1.3.0 ubo-gui-0.13.14 werkzeug-3.1.3 wsproto-1.2.0 yarl-1.19.0
+ sed -i '/^Banner /d' /etc/ssh/sshd_config.d/rename_user.conf
sed: can't read /etc/ssh/sshd_config.d/rename_user.conf: No such file or directory
```

@sassanh seems like there's a dependency issue. Have you seen this one?

zytegalaxy, 2025-04-12:
@sassanh Also the following file does not exist. 

```
# Remove the Raspberry Pi's SSH daemon banner warning about setting a valid user
sed -i '/^Banner /d' /etc/ssh/sshd_config.d/rename_user.conf
```

sassanh, 2025-04-14:
> [@sassanh](https://github.com/sassanh) Also the following file does not exist.

We need to make it ignore this line if the file doesn't exist. It may be that this file does not exist for the desktop version.

commit def28d83 (2025-05-07): fix(installation): gracefully handle missing file in sed command with || true - closes #281
commit d5611a25 (2025-05-08): fix(installation): gracefully handle missing file in sed command with || true - closes #281
closed 2025-05-08 by sassanh