ubopod/ubo_app #282: Failure to launch with new release image

nubcore, 2025-04-12:
Using the rpi imager with current release (1.3.0) build to create new microSD. Ring circles white while booting, then left at command line. Ubo screen never loads with or displays anything. Running and update and upgrade, additional packages are installed and the display renders. On reboot if left sitting at auto logged in console, something happens after a few moments and the hdmi output is changed to an extremely low resolution and the ubo screen graphics get inappropriately scaled. Device appears entirely unasable with a visible cursor that doesn't move. Hostname is overwritten to _ubo-bj_ for some reason as well on every reboot.

Using the raspberry pi configuration utility and changing from CLI to Desktop on boot seems to enable a usable environment. This auto logins the ubo account, even if auto login deselected, and can not log out to switch to different user.

Alternative work around, type _startx_ as soon as command prompt appears on boot.

[image]

[image]

[image]

zytegalaxy, 2025-04-12:
> Using the rpi imager with current release (1.3.0) build to create new microSD. Ring circles white while booting, then left at command line. Ubo screen never loads with or displays anything. 

I believe you downloaded the lite image correct? 

> Running and update and upgrade, additional packages are installed and the display renders. On reboot if left sitting at auto logged in console, something happens after a few moments and the hdmi [...]

When the device/app was booting up, did you have the monitor connected with HDMI? There's a known issue (https://github.com/ubopod/ubo_app/issues/137
) we are trying to address on the lite version that messes up with display if a monitor is connected to device as the app is loaded. 

If the monitor is connected AFTER the app is loaded, you are not going to see the issue in the first picture you sent. 

> Hostname is overwritten to _ubo-bj_ for some reason as well on every reboot.

We change the hostname name to `ubo-xx` intentionally so that each device has a unique hostname for accessing it via ssh or web UI. Do you have a preference on the name change? This just needs to be a unique name.

> Using the raspberry pi configuration utility and changing from CLI to Desktop on boot seems to enable a usable environment. 

The above issue only happens when there is no Desktop environment is present. On full image or after installing Desktop, the GUI is always displayed on the LCD display.

> This auto logins the ubo account, even if auto login deselected, and can not log out to switch to different user.

We force auto login as a work around for silencing screen reader for headless operation (no keyboard/mouse/monitor). We made this decision a long time ago but may revisit it. You should be able to logout and login with another user though. Do you have another user present on the system? 

Please note that you can add/delete users or reset user passwords using the GUI by going to Settings -> System -> Users

> Alternative work around, type _startx_ as soon as command prompt appears on boot.
> 
> [image]
> 
> [image]
> 
> [image]

zytegalaxy, 2025-04-12:
@nubcore On lite image, I installed Desktop using GUI by going to `Settings->System->Desktop` and was able to get Desktop environment (auto logged in as `ubo` user). I was also able to log out successfully. Please note that if you want to login back with user ubo, you have set a password for ubo user by going to `Settings -> System -> Users -> ubo -> Reset Password` or via do it via CLI. 

The logout issue might be related to the way you installed the Desktop environment since Raspberry Pi has moved away fro `X` and now used `Wayland`. 

When you click on Shutdown options, do you see this window:

[image]

BTW, this is where we set the autologin option:
https://github.com/ubopod/ubo_app/blob/dfdcdb1e7f70aead1084d2d5e0149e74ee5b4972/ubo_app/system/system_manager/package.py#L34

zytegalaxy, 2025-04-13:
@nubcore I was able to reproduce the logout issue on the full image (which has desktop installed). Can you please confirm if you downloaded the lite or full image release version? The full version has two parts that must be concatenated to make the full image with this command:  
`cat [[filename]]_* > [[filename]]`

I also ran into an issue of having to run `apt upgrade` before the app started working again.