ubopod/ubo_app #306: audio does not work until aplay is run by root

sassanh, 2025-05-22:
This is a draft of an issue until we have time to investigate it further

aplay before and after running the same command with root:
```sh
ubo@ubo-gp:~ $ aplay /opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/000-audio/sounds/done.wav
ALSA lib pcm_direct.c:2179:(_snd_pcm_direct_new) unable to create IPC semaphore
aplay: main:831: audio open error: Permission denied
ubo@ubo-gp:~ $ aplay /opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/000-audio/sounds/done.wav
Playing WAVE '/opt/ubo/env/lib/python3.11/site-packages/ubo_app/services/000-audio/sounds/done.wav' : Signed 16 bit Lit
tle Endian, Rate 16000 Hz, Mono
```

output of `sudo /usr/bin/wm8960-soundcard `:
```
+ dtparam i2c_arm=on
+ modprobe i2c-dev
+ sleep 5
+ for loop in 1 2 3 4 5
++ i2cdetect -y 1 0x1a 0x1a
++ egrep '(1a|UU)'
++ awk '{print $2}'
+ is_1a=UU
+ '[' xUU '!=' x ']'
+ echo 'install wm8960-soundcard'
install wm8960-soundcard
+ dtoverlay wm8960-soundcard
* Failed to apply overlay '5_wm8960-soundcard' (kernel)
+ sleep 1
+ rm /etc/asound.conf
+ rm /var/lib/alsa/asound.state
+ echo 'create wm8960-soundcard configure file'
create wm8960-soundcard configure file
+ ln -s /etc/wm8960-soundcard/asound.conf /etc/asound.conf
+ echo 'create wm8960-soundcard status file'
create wm8960-soundcard status file
+ ln -s /etc/wm8960-soundcard/wm8960_asound.state /var/lib/alsa/asound.state
+ break
+ alsactl restore
alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:1 use case configuration -2
No state is present for card vc4hdmi0
alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:1 use case configuration -2
Found hardware: "vc4-hdmi" "" "" "" ""
Hardware is initialized using a generic method
No state is present for card vc4hdmi0
alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:2 use case configuration -2
No state is present for card vc4hdmi1
alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:2 use case configuration -2
Found hardware: "vc4-hdmi" "" "" "" ""
Hardware is initialized using a generic method
No state is present for card vc4hdmi1
```

sassanh, 2025-05-23:
Reporting findings:
While trying to investigate why this is happening:
https://github.com/ubopod/ubo_app/actions/runs/15181455375/job/42703555633
I noticed this is happening on the old Pi4 runner ubo-fc:
[image]

fcorthay, 2026-10-25:
Check if `ubo` is in the `audio` group.

sassanh, 2026-10-25:
It should be as it's intended here: https://github.com/ubopod/ubo_app/blob/main/ubo_app/system/scripts/install.sh#L192