ubopod/ubo_app #310: Abstract Speech Recognition

sassanh, 2025-06-01:
We already have VOSK as our speech recognition but we need to abstract it so that future speech recognitions can be added by implementing this abstract class.

zytegalaxy, 2025-06-01:
@sassanh I don't know if you have considered using https://pypi.org/project/SpeechRecognition/ library. I also has that level of abstraction and VOSK API. I am not sure if their VOSK API does what you need though.

sassanh, 2025-06-01:
@zytegalaxy 
One other issue was that in our abstraction we need to separate word-detection with speech-recognition, limit the engine to particular grammer when it is waiting for wake word, doing wake word detection with one engine and speech-recognition with another, etc.

commit a5403d35 (2025-06-01): refactor(speech-recognition): abstract the speech recognition engine interface with `vosk` as the first implementation of it - close #310
commit 9a83b1f0 (2025-06-02): refactor(speech-recognition): abstract the speech recognition engine interface with `vosk` as the first implementation of it - close #310
commit d369d9d8 (2025-06-02): refactor(speech-recognition): abstract the speech recognition engine interface with `vosk` as the first implementation of it - close #310
commit 6fe3fac4 (2025-06-02): refactor(speech-recognition): abstract the speech recognition engine interface with `vosk` as the first implementation of it - close #310
closed 2025-07-10 by sassanh